petrobras/ross #2: Define project license

raphaeltimbo, 2018-12-18:
The project license should be permissive considering the use of the library in the industry.

rodrigomoliveira1, 2018-12-19:
[software_license.txt](https://github.com/ross-rotordynamics/ross/files/2695633/software_license.txt)

There are lots of licenses models we could use to build our project. The link above leads to a file which summarizes the main types of licenses commonly used in the open-source softwares area. They are described in the order of the strongest model for the weakest. It means that the strongest models are more tied to the conditions (more conditions) than the weakest (fewer conditions). Anyway, it doesn't mean the strongest models are better for this software.

In fact I suggest picking the MIT License, once it has the same permissions like the others and we are not running for a patent (the GNUs licenses grants patents permission for the contributors), and we keep it simple like the MIT License is.

I hope it was useful! What do you think about it?

raphaeltimbo, 2018-12-20:
@rodrigomoliveira1 , I agree with your suggestion of picking the MIT License. 
Additionally, I think we should use this issue to track all dependencies that we may include in the project and check if we have any conflicts with our license.
For now, I believe we are going to use the following packages (their licenses are linked):

- [numpy](https://github.com/numpy/numpy/blob/master/LICENSE.txt)
- [scipy](https://github.com/scipy/scipy/blob/master/LICENSE.txt)
- [matplotlib](https://github.com/matplotlib/matplotlib/blob/master/LICENSE/LICENSE)
- [bokeh](https://github.com/bokeh/bokeh/blob/master/LICENSE.txt)
- [pytest](https://github.com/pytest-dev/pytest/blob/master/LICENSE)

Looking at their licenses, I do not see any conflict with the MIT License.

rodrigomoliveira1, 2018-12-20:
I have just checked the licenses of the packages that you listed above, and i don't see any conflict either. They're using 2-clause BSD, 3-clause BSD or MIT License, which are pretty the same thing.
All of them allows us to use each package privately or commercially.

And yes, as we add packages to our software, we could list the respective licenses for each package, just as do the developers of those programs that you listed.

Just one more thing: in case we need, eventually, change our license for some kind of random reason, is it possible to do or once we have estabilished a license it's permanent?

raphaeltimbo, 2018-12-20:
I do not think it is permanent. The biggest problem would be if we wanted to change from MIT to GPL. In that case I think we would have versions of the code with different licenses (matplotlib for example has a different license for versions prior to 1.3.0). But I don't think we will have this kind of problem.
For minor changes to the license I think we are OK. It is possible to see that the licenses for the packages I have listed have been modified (you can see that on "blame").

commit 15d036a3 (2019-01-11): Add LICENSE.

This adds the MIT License and closes #2.
closed 2019-01-11 by raphaeltimbo