petrobras/ross #3: Create document to detail contribution process

raphaeltimbo, 2018-12-18:
Write a contribution guide, detailing cloning, commit, pull requests etc.
Take as an example https://docs.scipy.org/doc/numpy/dev/

flaviorangel, 2018-12-23:
Taking NumPy's manual as reference, since it is a largely applied library and well respected repository in GitHub, we should begin by introducing to the very basics of how to install Git before getting to the development workflow. For now, we will focus on the development workflow rather than explaining the basics. That said, this should be step 1 (I will write one comment for each step here):

1. Make you own copy (fork) of ROSS:
git clone https://github.com/ross-rotordynamics/ross.git
cd ross
git remote add upstream https://github.com/ross-rotordynamics/ross.git

A few comments:
1. This sequence of commands was copied from NumPy's manual. 
2. I still must test it to make sure it creates this 'ross' folder. I was not with access to a terminal with Git when wrote this part.
3. Any name can be given to this remote. The manual describes "upstream" as "just the arbitrary name we’re using to refer to the main NumPy repository at NumPy github". We can either keep it or change to something similar to "[shortname]", as written in Git's documentation*.

* Access: https://git-scm.com/book/it/v2/Git-Basics-Working-with-Remotes

flaviorangel, 2018-12-23:
NumPy manual suggests the following commands to keep in sync with changes in NumPy (maybe this should be our step 2):

2. Keep in sync with changes in Ross
git config branch.master.remote upstream
git config branch.master.merge refs/heads/master

A few comments:
1. It states that this is a way to set up your repository so it pulls from "upstream" by default. I still need to test these commands to make sure they work with our repository.
2. I'm assuming that with these commands "git pull" pulls from the remote "upstream", but it could be replaced by other commands that specify the repository to pull from, such as "git fetch upstream". I must still test them.

flaviorangel, 2018-12-23:
After some changes are made, described in NumPy manual as "a complete, working set of related changes", step 4 should be performed:

4. Push changes to your git repository:
git add modified_file
git commit
git push origin my-new-feature

Comments:
1. According to NumPy manual, a text editor will open up to allow a commit message to be written. The manual explains how a proper message should be*. We should give similar instructions, although this is not priority right now.

*Check: https://docs.scipy.org/doc/numpy/dev/gitwash/development_workflow.html#writing-the-commit-message

flaviorangel, 2018-12-23:
Finally, for contributors to create a pull request (PR), NumPy refers to Github's web page on PRs: https://help.github.com/articles/about-pull-requests/

If changes involve modifications to the API or addition/modification of a function, it refers to a mailing list. Maybe we should keep one as well.

As for core developers, the final step should be:

5. Pushing changes to the main repo:
First, it merges or rebases on the target branch, then check that what is being pushed looks sensible. Finally, pushes to upstream.
git fetch upstream
git rebase upstream/master
git log -p upstream/master..
git log --oneline --graph
git push upstream my-feature-branch:master

Comments:
1. The first two commands should be used if there are only a few, unrelated commits. If it is not the case (i.e, if all commits are related), they should be replaced by:
git fetch upstream
git merge --no-ff upstream/master
2. NumPy manual does not refer to a flow between core developers to perform a check on each other's pushs to the main repo. We initially thought we should maintain such routine, but maybe it is not necessary.

raphaeltimbo, 2018-12-26:
This is great @flaviorangel. 
I have added the following file to the project: `ross/doc/git_configuration.rst`.
Could you please edit this file copying and pasting what you have posted here and then submit a pull request.

flaviorangel, 2019-01-09:
As we discussed in our meeting, our references for this file will be: 
https://github.com/numpy/numpy/blob/master/doc/source/dev/gitwash/git_intro.rst
http://www.sphinx-doc.org/en/master/usage/restructuredtext/basics.html

commit 0a022ae3 (2019-01-14): Add instructions to fork to user account before cloning

Closes #3
closed 2019-01-14 by raphaeltimbo