petrobras/ross #5: Create objects, attributes, methods and UML diagram

raphaeltimbo, 2018-12-18:
Create objects, attributes, methods and UML diagram.

GabrielBachmann, 2018-12-23:
I also included the original XML file that can be edited in case there is any change of course in the project's design, and a HTML version, on the google drive project folder.

[image]

raphaeltimbo, 2018-12-26:
@GabrielBachmann, this is looking good. What tools did you use to build the diagram?
I think we can put the .xml file in the `ross/doc/` folder for now. Could you add the file and open a pull request?

GabrielBachmann, 2018-12-26:
I've put the .xml file in the doc folder. The tool used to build this diagram is avaible on www.draw.io
The license terms were checked, as you can see in this link: 
https://desk.draw.io/support/solutions/articles/16000042494-what-are-the-usage-terms-for-diagrams-produced-by-draw-io-

raphaeltimbo, 2019-03-20:
I will be closing this issue for now. We can reopen it if needed.

closed 2019-03-20 by raphaeltimbo