petrobras/ross #13: Build documentation using sphinx

raphaeltimbo, 2019-01-09:
Implement documentation using sphinx.

GabrielBachmann, 2019-01-14:
https://www.youtube.com/watch?v=QNHM7q2hLh8

flaviorangel, 2019-02-18:
Github pages has been activated. Now I must push the html files to a 'docs' folder in the root in order to get the webpage to work. Since the old branch was merged to master, I will create a new one related to this issue

flaviorangel, 2019-02-25:
Refer to oct2py github page

flaviorangel, 2019-02-26:
I got gh-pages working on my repository. Now I'll create a PR and close this issue. We need to think of a new name for our 'doc' folder, as we will have a new 'docs' folder from now on. Or maybe we can resort to the gh-pages branch strategy.

For future reference, here is the problem I was facing and the solution I found: I didn't know what exactly should be in the 'docs' folder in order to create the gh-page, and I would always get a 404 error. It turns out the 'docs' folder should include the content of the 'html' folder created by sphinx; everything else can be ignored. I had tried this solution already, but some files were missing without me noticing. The reason was they were not being included by 'git add -f *' command (-f option is to include ignored files and * is to include everything). I realized this when I compared my 'docs' folder with others (or gh-pages branches) from pages on github using sphinx to generate them. The missing files were '.nojekyll' and '.buildinfo'.

PR #51 (2019-02-26): Include docs folder and necessary html files
Close #13
closed 2019-02-26 by rodrigomoliveira1