petrobras/ross #18: Issue with line endings

raphaeltimbo, 2019-01-15:
In the #8 PR I have noticed that there was a ^M character at the end of some lines:
[image]

It looks like this breaks the diff for the file.

rodrigomoliveira1, 2019-01-15:
I just found an explaination for this problem: 

"_The ^M is a carriage-return character. If you see this, you're probably looking at a file that originated in the DOS/Windows world, where an end-of-line is marked by a carriage return/newline pair, whereas in the Unix world, end-of-line is marked by a single newline._"

So, i'm removing the line break and leting all the sentence in a same line to avoid it.

raphaeltimbo, 2019-01-15:
There is also a problem related to line endings on different platforms.
[This article](http://web.archive.org/web/20150912185006/http://adaptivepatchwork.com:80/2012/03/01/mind-the-end-of-your-line/) explains the problem and how to configure git to take care of end of line normalization.

I will add a .gitattributes file to the repository to deal with this issue.

commit 48f2dd6a (2019-01-15): Add .gitattributes file.

Add .gitattributes with the * text=auto option to normalize the text
files line endings to LF and avoid problems between linux/osx and
windows users. It also avoids problems with git diffs.
This is explained in details here:
http://web.archive.org/web/20150912185006/http://adaptivepatchwork.com:80/2012/03/01/mind-the-end-of-your-line/

This closes #18.
closed 2019-01-15 by raphaeltimbo