petrobras/ross #19: Class element is repeated on different modules

raphaeltimbo, 2019-01-17:
The `class Element` is repeated on different files: shaft_element.py, disk_element.py...
I think we should have that class in a separate element.py file and the other modules should import from it.

rodrigomoliveira1, 2019-01-17:
Sounds good! i'm doing it right now. Just moved the `class Element` to a new element.py and removed it from the orignials ones.

rodrigomoliveira1, 2019-01-17:
A new file element.py has been added. The commit is [here](https://github.com/ross-rotordynamics/ross/pull/20/commits/e53c436a86118c414b1ce3ebe3eab64b9bab6945)

rodrigomoliveira1, 2019-01-18:
So, I just commited changes in 3 files removing the `class element` from them. They're listed below:

- [shaft_element.py](https://github.com/ross-rotordynamics/ross/commit/bc534ec07c4eaba36f156cc145efc80379bf7b0d)

- [disk_element.py](https://github.com/ross-rotordynamics/ross/commit/330745446bc27c518e669333657a7adf4e31107d)

- [bearing_seal_element.py](https://github.com/ross-rotordynamics/ross/commit/5f2a75da7b2ac31f59cf3d76475d3786ac6d4009)

They are all together in a new branch called "element_file" including the element.py file

closed 2019-01-21 by raphaeltimbo