petrobras/ross #34: Convergence analysis

rodrigomoliveira1, 2019-02-11:
Since we've already started working on tests to generate some results, I thought it would be useful to introduce a pre-processing test. It's a convergence analysis. It refines the shaft elements mesh and checks the relative error between the eigenvalues (self.evalues) and the undamped natural frequency (self.wn) calculated in each step of refinement, and it stops when the relative error reaches a particular value.

rodrigomoliveira1, 2019-02-11:
[rotor_convergence.py](https://github.com/ross-rotordynamics/ross/commit/8a4b7a5da581df142061519104e12f1947032934)

This is the first version. The docstring provides further information about the input / output data.

This code also introduce a different way to input the rotor variables. Instead of entering the elements itself, the user may now divides the rotor in regions (as many you want).
For example: if we got a shaft that has 3 sections with different diameters, we could divide the shaft in 3 regions (it could be more regions).

The user defines the number of regions, lenght and diameter. The number of elements per regions will be calculated.

So, why that? Because it spares time to build the model and make things more intuitive. It takes longer to enter a 100 element' model than a 10 regions rotor with 10 elements each.
For now, all the regions will have the same number of elements, and if a region has much larger elements than others, the solution is divide this particular region in more regions.

If you like this way, we could adapt it better and for now on, use this method to input the rotor's data.

rodrigomoliveira1, 2019-02-11:
-> TODO

- Return the convergence analysis graphs: **Relative Error vs Number of Elements** and **Natural Frequency vs Number of Elements**

- Improve the elements distribution per region. Different number of elements per region.

- Build elements with different lenghts.

rodrigomoliveira1, 2019-02-18:
-> TODO 

- Return the convergence analysis graphs. (check)

- Create a classmethod to build the rotor with sections (or regions) in rotor_assembly.py.

- The class convergence should be a method only, instead of a class.

rodrigomoliveira1, 2019-02-20:
-> TODO

- Return the convergence analysis graphs. (check)

- Create a classmethod to build the rotor with sections (or regions) in rotor_assembly.py. (check)

- The class convergence should be a method only, instead of a class. (check)

All the changes marked as "check" are done in [rotor.assembly.py](https://github.com/ross-rotordynamics/ross/commit/474a0f810a90e4d31b93e888fac3a00d48e76e65). 
This commit has been uploaded in rotor-analysis branch. There is a need for adding pytest codes to check these analysis yet. Besides, the rotor_convergence file will be deleted once the tests are done.

raphaeltimbo, 2019-03-20:
Closed by PR #52

closed 2019-03-20 by raphaeltimbo