petrobras/ross #35: Fluid flow code - #TODO

rodrigomoliveira1, 2019-02-14:
[Fluid_flow.py](https://github.com/ross-rotordynamics/ross/commit/6306bbd05afa3d154cfc4bc143dad4cee9b90327) was added in a new branch called "fluid-flow".
Here are some changes that should be made in it.

- Translate comments, names of variables and functions.

- Add docstring to explain functions.

- check PEP 8

- Remove TODO from the code and add them here

flaviorangel, 2019-02-18:
Comments and names have been translated. The remaining tasks are still pending.

flaviorangel, 2019-02-25:
These TODOs are commented in the [fluid flow branch](https://github.com/ross-rotordynamics/ross/tree/fluid-flow). They are being solved gradually, but this issue won't be closed until this branch is ready to merge, which will depend mostly on the [example test issue](https://github.com/ross-rotordynamics/ross/issues/37).

JuliaMota, 2019-03-06:
I checked PEP8 and now it's ok.
Correct documentation details missing.

flaviorangel, 2019-03-08:
I just pushed changes to the entire fluid-flow code, including those related to these TODOs. I ended up adding a new one: the user needs to be able to choose the structure of the internal radius. I think that belongs to a new issue. This one can be closed as soon as the merges are pushed to the master.

JuliaMota, 2019-03-14:
Some graphics need proper titles and subtitles

commit 54f9c8c6 (2019-03-27): Fix plot and add titles to plots

This commit make the fluid flow routines ready to use for short bearings
and seals. The changes we made along the last commits also solve TODOs
mentioned in issue 35. Close #35
closed 2019-03-27 by raphaeltimbo