petrobras/ross #37: Add book examples to fluid flow code and verify results

flaviorangel, 2019-02-18:
We need to add book examples to fluid flow code to the code instead of the entry data we have now. We also need to verify if the results are according to expected.

JuliaMota, 2019-02-27:
I'm searching for examples that return the same information as our code. I've found some things that might be interesting in the book TRIBOLOGY SERIES, but these still need to be understood and adapted. 

Initially, we can include the density and viscosity of the ISO VG 32 and 46 oils to analyze the pressure behavior.

- Viscosity (Pa.s):
#visc= 0.032 # ISO VG 32 (40°C)
#visc= 0.0054 # ISO VG 32 (100°C)
#visc= 0.046 # ISO VG 46 (40°C)
#visc= 0.0068 # ISO VG 46 (100°C)

- Fluid density (Kg/m^3):
#rho = 857. # ISO VG 32
#rho = 861. # ISO VG 46

JuliaMota, 2019-03-06:
The examples of viscosity and density quoted were included. But I have difficulty adapting the examples from the book to the code.In the book TRIBOLOGY SERIES, there is an example of an annular seal that is very close to our case. However, there seems to be a lack of necessary information. I keep researching and working on it.

flaviorangel, 2019-03-08:
> The examples of viscosity and density quoted were included. But I have difficulty adapting the examples from the book to the code.In the book TRIBOLOGY SERIES, there is an example of an annular seal [...]

This example doesn't seem to fit our case. Although it gives all entry data we need, except for grid information, its outputs are axial leakage rate of flow, friction torque, and dissipated power. Unless we can calculate those using our pressure matrix, it can't be adapted. It also infers the pressure as p = (1 - z/L), which are able to infer as well given a simple structure.

JuliaMota, 2019-03-14:
Me and @flaviorangel  implemented an example from chapter 5 of the book "Tribology Series 33". The book presents the Sommerfeld solution for pressure calculation. We produced a graph that compares the analytical solution presented in the book with the results presented in our code. However, the results are different. See the graph: the red line represents the analytical result of the book while the blue line represents our result.
[image]
Studying the text better, I have identified some possible errors:
1- The example shows a static case, so we should consider omega = 0
2- The book presents more than one static case. The case that we implemented is a long bearing (page 118), ie, according to the book L/D > 4 (L and D are width and diameter, respectively).

Note: An example of a short bearing is shown on page 127. I believe we can test it.

flaviorangel, 2019-03-18:
Here what needs to be done to close this issue:

- [ ] Add analytical expression to the code in order to compare results;
- [ ] Add tests to integrate with pytest;
- [ ] Add a working example from a book with results matching with ours.

Anything else?

flaviorangel, 2019-03-25:
We need to specify where our 'zero' is located in the circle. Some books suggest that it begins where 'h' is shorter, 'h' meaning the space between the rotor and the stator.

raphaeltimbo, 2019-06-28:
I think this issue involves lots of different things. 
Maybe we could break it into different smaller issues (and consequently different PRs) to make it easier to track the work?

flaviorangel, 2019-07-01:
> I think this issue involves lots of different things.
> Maybe we could break it into different smaller issues (and consequently different PRs) to make it easier to track the work?

Agreed. I'll break it in three and close this one.

flaviorangel, 2019-07-14:
This issue was broken into issues #201 and #202. As for the short bearing, by the time it was broken, we were already being able to calculate its pressure.

closed 2019-07-14 by flaviorangel