petrobras/ross #45: Implement rotor plot with bokeh

raphaeltimbo, 2019-02-20:
Currently, the rotor is plotted using matplotlib, but it would be nice to have a plot using the [bokeh library ](https://bokeh.pydata.org/en/latest/).
With this library, we can create interactive plots for web browsers. It would be nice to have a hover tool to show the properties of each rotor part.
@rodrigomoliveira1 @flaviorangel @GabrielBachmann can one of you guys take a look on this?

rodrigomoliveira1, 2019-02-22:
I have read about bokeh since then. I think it's a good idea to migrate to bokeh library. These interactive plots seems easy to manage. 
Creating simple graphs like in matplotlib is simple as in this, but running an interactive plot isn't. We need to start a bokeh server that holds the data, which will be accessed by the interactive plot.

So, i have been learning how to create bokeh servers. I think we have 2 main options:

- a local or individual use Bokeh server: we create an app and we are able to share it between ourselves;

- a Bokeh server to publish interactive data visualizations and applications that can be viewed and used by a wider audience.

I successfully tested a local server using the [bokeh user guide](https://bokeh.pydata.org/en/latest/docs/user_guide/server.html#userguide-server-applications], although I needed to use the command prompt line to start it. I couldn't access it only by a python command line. It opens a chrome/firefox/IE window showing the graph and you can use the interactive tools there.

rodrigomoliveira1, 2019-03-13:
The fisrt enhancements is followed by [PR#63](https://github.com/ross-rotordynamics/ross/pull/63).

rodrigomoliveira1, 2019-04-02:
TODO:
Finish adding bokeh plot to rotor_assembly.py / results.py plot methods:

- [ ] plot_rotor

- [ ] campbell

- [ ] freq_response

- [ ] forced_response

- [ ] mode_shapes

- [ ] plot_ucs

- [ ] plot_level1

- [ ] plot_time_response

- [ ] static

- [ ] convergence

rodrigomoliveira1, 2019-04-02:
I've been looking for some formats which bokeh is able to export to. 
Currently, bokeh can only export to .PNG and .SVG formats. PDF is not supported.

Both formats have some kind of limitation:

- PNG: Responsive sizing_modes may generate layouts with unexpected size and aspect ratios. It is recommended to use the default fixed sizing mode.
- SVG: It’s not possible to create a single SVG for a layout of plots, as each plot is it’s own distinct SVG element. Alternatively, it’s possible to download a SVG plot using a SaveTool from the toolbar. However, currently an SVG export of a plot with a toolbar will have a blank area where the toolbar was rendered in the browser.

rodrigomoliveira1, 2019-05-21:
Bokeh plot is now available in all plot methods in rotor_assembly.py, with exception "mode_shapes", because bokeh plot does not have a module to do a 3D plot yet.

rodrigomoliveira1, 2019-05-25:
Pull Request #121 have been merged and closes this issue.

closed 2019-05-25 by rodrigomoliveira1