petrobras/ross #48: Change fluid flow plots to bokeh library

flaviorangel, 2019-02-25:
Fluid flow is currently being plotted using matplotlib. We should change to bokeh library. However, fluid flow is still a branch and we still haven't fixed some of its issues, such as adding an example from literature and make sure results are matching. We should focus on solving previous issues regarding fluid flow before focusing on this issue.

flaviorangel, 2019-04-07:
This issue is on hold until we are sure which library we will use for plotting. Although bokeh is better to interact with, we know matplotlib is good for adding to reports. Is bokeh also good for reports? I'm not sure of who was put in charge of comparing them, but we should have some update soon.

flaviorangel, 2019-04-08:
It was decided that we will create new plots using bokeh, but keep the ones we have in matplotlib.

flaviorangel, 2019-04-18:
It seems bokeh doesn't have a method for plots in polar coordinates. Our cylindrical plot of the pressure along theta will have to be on cartesian coordinates.

raphaeltimbo, 2019-04-18:
There is an issue open [here](https://github.com/bokeh/bokeh/issues/657) about this.
It suggests the following code:
```python
from math import pi

from bokeh.io import show
from bokeh.models import ColumnDataSource, CustomJSTransform
from bokeh.plotting import figure
from bokeh.transform import transform

p = figure()

source = ColumnDataSource(data={
    'color' : ['red', 'blue'] * 20,
    'r'     : [1, 2]*20,
    'theta' : [2*pi/360 * x for x in range(0, 360, 9)],
})

polarx = CustomJSTransform(args=dict(source=source), v_func='''
const new_xs = new Array(source.data.r.length)
for(var i = 0; i < new_xs.length; i++) {
    new_xs[i] = source.data.r[i] * Math.sin(source.data.theta[i] )
}
return new_xs
''')

polary = CustomJSTransform(args=dict(source=source), v_func='''
const new_ys = new Array(source.data.r.length)
for(var i = 0; i < new_ys.length; i++) {
    new_ys[i] = source.data.r[i] * Math.cos(source.data.theta[i] )
}
return new_ys
''')

p.circle(x=transform('r', polarx), y=transform('r', polary),
         color='color', size=15, alpha=0.8, source=source)

show(p)
```
Although it is not a simple solution, I think we could use something like that.
If it becomes too complicated, I suggest keeping only matplotlib for now.

flaviorangel, 2019-04-25:
I could not find a way to plot the way we are looking for. The colors would have to be represented by numbers instead of strings in order to show the pressure gradually changing. We can go back to this issue in the future, but for now matplot is doing okay in the cylindrical plot.

PR #94 (2019-05-03): Add bokeh plot to fluid flow
Add routines to fluid flow to plot using bokeh instead of matplotlib. The previous routines were kept, but their names were changed to "matplot" instead of "plot". Close #48
closed 2019-05-03 by raphaeltimbo