petrobras/ross #53: Specify programs being used

flaviorangel, 2019-02-27:
This was a suggestion made by @thiagoritto, so that newcomers to the project (or online collaborators) can know straight away which applications they should install and what they do.

raphaeltimbo, 2019-02-28:
One idea would be to have a section "Dependencies" on README.md.
There we could list the main dependencies and python versions that can be used.
But the user won't have to worry about installing all dependencies since pip will take care of this by looking at the packages listed on the setup.py file (`REQUIRED` variable).

flaviorangel, 2019-03-12:
What Ritto had in mind when suggesting this issue was more of a way to integrate people to the project than help them run the code. So it would explain applications like pycharm, git, and black more than numpy, matplotlib, etc, specially for people not used to them. But you're right: explaining in the readme that the dependencies are listed on setup.py and how to install them should also be included.

raphaeltimbo, 2019-03-13:
We could add a section to the documentation related to how to contribute to the project. 
The git configuration (that is already written) could be a subsection of it and we could also add an explanation about black and the other things on that section. 
What do you think?

flaviorangel, 2019-03-14:
I'll work on it.

PR #62 (2019-03-27): Update documentation and include introduction
Documentation was updated to look more clear and with more references.
An introduction.rst file was created. It explains the very first steps
to start using ROSS. Unlike the tutorial, which will guide through how
to run the program, introduction tells what the user needs to install
in order to get the program ready to run. Close #53
commit 80dcaf76 (2019-03-27): Update documentation and include introduction

Documentation was updated to look more clear and with more references.
An introduction.rst file was created. It explains the very first steps
to start using ROSS. Unlike the tutorial, which will guide through how
to run the program, introduction tells what the user needs to install
in order to get the program ready to run. Close #53
closed 2019-03-27 by raphaeltimbo