petrobras/ross #54: Documentation page not showing

raphaeltimbo, 2019-02-28:
The documentation page generated by sphinx is not showing on https://ross-rotordynamics.github.io/ross/.
For some reason the README.md is displayed on the page instead of the html generated when running sphinx.

flaviorangel, 2019-02-28:
Ross master seems to be identical to the [ross master branch in my repo](https://github.com/flaviorangel/ross). I don't understand though why it says "This branch is 2 commits ahead, 2 commits behind ross-rotordynamics:master" since the PR originated from there and was merged without changes (github bug maybe?). I clicked "Compare" to investigate: master and base are set correctly, but it shows as if it was comparing to a previous version of ross master, as if my 2 last commits were not merged...

Anyway, [the page generated from my repo](https://flaviorangel.github.io/ross/) is working fine, therefore the page from ross should be working fine as well. I assume this bug is maybe related to ross settings, so could you please take a look, @raphaeltimbo, and make sure the page is being built from the docs folder? If it is already, could you try changing to None and then back?

I tried to reproduce the error by changing the way the page is built in my repo. I changed to build from the master branch and, surprisingly, it built correctly. We might not need a docs folder after all. That would make things easier for whoever is using sphinx, as they wouldn't have to move files to the docs folder when making html.

raphaeltimbo, 2019-03-10:
@flaviorangel , it was actually an error on the repository's settings.
[image]

I have changed the source to the docs folder and now the correct front page is showing.

I will close this issue.

closed 2019-03-10 by raphaeltimbo