petrobras/ross #55: Add package to pypi

raphaeltimbo, 2019-02-28:
We will need to add the package to the pypi (python package index) to enable installation of the package with pip through the command:
`$ pip install <package name>`

The name ross is already taken on pypi, so we will need to think about an alternative.

flaviorangel, 2019-04-02:
[This official guide](https://packaging.python.org/tutorials/packaging-projects/) explains how to upload packages to pypi. It suggests to upload first to Test PyPI, which isn’t a permanent storage. I'll try this test first and then get everything ready to upload our code when we have a first release.

I was reading [this other guide](https://medium.com/@joel.barmettler/how-to-upload-your-python-package-to-pypi-65edc5fe9c56) earlier. It says the git repo needs to have the exactly same name as the package. If that is true, we might have to rename our repository at some point, as 'ross' is already taken. I will investigate.

flaviorangel, 2019-05-24:
I have been trying to follow the PyPi tutorial, but unsuccessfully. I'm getting an ImportError on the last step. I've searched for this same problem online, but no one seems to report it, so there must be something wrong with my computer. @raphaeltimbo can you try to execute these last steps, just so I make sure it works? An example package was already uploaded to TestPyPI. Please try the following command sequence:
`python3 -m pip install --index-url https://test.pypi.org/simple/ --no-deps example-pkg-flaviorangel
python`
`import example_pkg`
`example_pkg.name`

raphaeltimbo, 2019-05-25:
@flaviorangel , It is not working here:
[image]

flaviorangel, 2019-05-31:
We were able to solve this bug. I didn't follow the instructions correctly when I created the files: they were supposed to be in different folders. It worked later and then we successfully uploaded ross to PyPI. It can be found [here](https://pypi.org/project/ross-rotordynamics/). A user can now follow instructions in [ross documentation](https://ross-rotordynamics.github.io/ross-website/installation.html) to install it. This simple pip command will install ross and all its requirements:
`pip install ross-rotordynamics`
As "ross" was already taken in PyPI, we chose "ross-rotordynamics" instead. Regardless, in order to import ross to python, a simple `import ross` command is enough. This comes from the fact that the name of the main folder we uploaded to PyPI is "ross". The only problem will occur if a user installs more than one package named ross (such as the "ross" that came first in PyPI) and tries to import one of them. We are not sure of what will happen, but that is very unlikely since this other "ross" is related to a different field (regular expressions in bots).
The alpha release still depends on issues #84 and #132 . This last one was identified when testing ross installed through pip. I'll be working on it. Later today, I'll upload the most recent version to PyPI. I'm closing this issue.

closed 2019-05-31 by flaviorangel