petrobras/ross #56: Generate report for profiling

raphaeltimbo, 2019-02-28:
We need to develop a way to profile our code, measuring the time spent at each step of our calculations.
This can be done using [cProfile](https://docs.python.org/3/library/profile.html?highlight=cprofile#module-cProfile) and [snakeviz](https://jiffyclub.github.io/snakeviz/) for visualization.

The profile report will show something like the following (taken from the snakeviz page):

> [image]
>  
> <p>The table contains one row per unique function called.
> Calls to the same function from different places are all grouped into
> one row.
> The columns are the same as described in the
> <a href="https://docs.python.org/3.4/library/profile.html#instant-user-s-manual">cProfile user&rsquo;s manual</a>:</p>
> <ul>
> <li><strong>ncalls:</strong> Total number of calls to the function.
>     If there are two numbers, that means the function recursed and
>     the first is the total number of calls
>     and the second is the number of primitive (non-recursive) calls.</li>
> <li><strong>tottime:</strong> Total time spent in the function,
>     not including time spent in calls to sub-functions</li>
> <li><strong>percall:</strong> <code>tottime</code> divided by <code>ncalls</code></li>
> <li><strong>cumtime:</strong> Cumulative time spent in this function and all sub-functions</li>
> <li><strong>percall:</strong> <code>cumtime</code> divided by <code>ncalls</code></li>
> <li><strong>filename:lineno(function):</strong> File name and line number were the
>     function is defined, and the function&rsquo;s name</li>
> </ul>

With this report we will be able to:

- Know which functions are slow and need optimization;
- Check the impact of an optimization and see if it is worth doing it (considering that some optimizations can make the code less readable and be harder to maintain);
- Check if some new feature is making our code slow;
- Have benchmarks for each realease.

raphaeltimbo, 2019-04-10:
One other point that I would like to add to this issue is the import time of the library.
It looks to me that it is taking a lot of time to make the import and maybe we can make some optimization on that.

GabrielBachmann, 2019-05-31:
#126 closes this issue.

closed 2019-05-31 by GabrielBachmann