petrobras/ross #59: Create choices in internal radius function in fluid flow and make them user friendly

flaviorangel, 2019-03-08:
The rotor radius is being calculated arbitrarily. The user must be able to choose which structure they want for the rotor radius. This choice should be done when the object is instantiated in the form of a parameter the user will give, if they so wish.

raphaeltimbo, 2019-03-10:
Hi @flaviorangel !
Could you give more details about the possible structures that the user can choose?

flaviorangel, 2019-03-12:
By structure I meant geometry of the rotor. The geometries presented in Luiz Queiroz's undergraduate dissertation are cylindrical, sinusoidal, conical, and square. When instantiating the class, the user should be able to choose among geometries. A more advanced version could let the user give their own customized geometry as a parameter.

raphaeltimbo, 2019-03-13:
Got it! But I don't know if all these geometries would make sense for our case. 
I remember that the dissertation that you have cited had some parts related to drilling, and I am wondering if these different geometries would not be more related to that context.
Maybe we should check if all those options are really needed.

flaviorangel, 2019-03-14:
I agree. Let us discuss the issue in our next meeting.

flaviorangel, 2019-04-07:
This issue should be closed and reopened when it is a better moment to be implemented, since it is an advanced feature and don't relate to the tests we are doing now. We are lacking a label to identify such issues.

raphaeltimbo, 2019-04-08:
I have just created a label "wontfix for now".

closed 2019-04-08 by raphaeltimbo