petrobras/ross #75: Make notebooks with Friswell examples

raphaeltimbo, 2019-04-01:
I think we should take Friswell examples and exercises and reproduce them with ross.
These notebooks can be added as part of the documentation.

- [ ] Example 5.8.1
- [ ] Example 5.8.2
- [ ] Example 5.9.1
- [ ] Example 5.9.2
- [ ] Example 5.9.4
- [ ] Example 5.9.5
- [ ] Example 5.9.6
- [ ] Example 5.9.7
- [ ] Example 5.9.8
- [ ] Example 5.9.9
- [ ] Example 5.9.10
- [ ] Example 6.2.1
- [ ] Example 6.2.2
- [ ] Example 6.2.3
- [ ] Example 6.3.1
- [ ] Example 6.3.2

GabrielBachmann, 2019-07-14:
With the current state of the code, we can't make some of these examples, here is a list of them:

- Example 5.9.8 (4 pole machine)
- Example 6.2.1(mass eccentricity)
- Example 6.2.2(disk is misaligned with the axis of the shaft by 1°)
- Example 6.3.2(Bent shaft)
- Example 6.3.1(Out-of-balance disk)

Besides that, I've read again the Example 6.2.3(Unbalance response) and I think this one can be made.

GabrielBachmann, 2019-11-11:
I'm closing this issue since we don't have interest in making the example 6.2.3 right now.

closed 2019-11-11 by GabrielBachmann