petrobras/ross #77: Create automatic report

raphaeltimbo, 2019-04-01:
Generate a report with the main results from the rotordynamic analysis.
Some criteria should be used to state if the report is ok according to a standard.

rodrigomoliveira1, 2019-06-18:
Thinking about it, maybe we could create a new file to do this analysis according to the standards and then generate a report. But first of all, we should work in this new file, what you think?

raphaeltimbo, 2019-06-20:
I agree that creating a file to do the analysis is the first step.
We can worry about creating the actual report later. For reference, I saw [this blog post](https://pbpython.com/pdf-reports.html).

raphaeltimbo, 2020-01-21:
We have a PR open to solve this issue (#427).

rodrigomoliveira1, 2020-03-10:
PR #427 has been merged and closes this issue.

closed 2020-03-10 by rodrigomoliveira1