petrobras/ross #79: Allow inputs from excel

flaviorangel, 2019-04-02:
The class BearingElement must allow inputs from excel type files. A separated routine should be included to convert from excel to csv and then from csv to toml.

raphaeltimbo, 2019-04-04:
@flaviorangel , you can find the xl files [here](https://github.com/raphaeltimbo/ross/tree/master/ross/tests/data).
I think you can convert from excel directly to toml.

flaviorangel, 2019-04-08:
> @flaviorangel , you can find the xl files [here](https://github.com/raphaeltimbo/ross/tree/master/ross/tests/data).
> I think you can convert from excel directly to toml.

Do the xl files need to follow a certain format or can the user define their own?

I wrote [this code](https://github.com/flaviorangel/ross/blob/input-from-file/ross/aux_routines.py) that searches in a xl file for a block of data with the given size and returns a numpy array containing only this block. This way the user can write the xl file as he or she prefers. The problem is not every argument needs to be specified when instantiating the BearingElement class, so the program needs to know which arguments were passed. I could add a pre-process routine that gets the names of the arguments given from the file (the user would need to specify them as column names), but it would be much easier if the xl file would follow a format. For example, first row with all the parameters of the BearingElement class and following rows containing the arguments. What do you think?

raphaeltimbo, 2019-04-08:
Yes, let's follow a format link you've suggested.

flaviorangel, 2019-04-11:
The BearingElement class contains this 'n' argument that is not explained in the docstring. I didn't understand what it does and it was not mentioned when we talked about the arguments that should be in each line of the excel table. Should there be a different 'n' per line of arguments? Or is it supposed to carry the number of lines?

raphaeltimbo, 2019-04-11:
The `n` argument is related to the node in which the bearing will be located.
I have an issue for that.
Regarding the excel import, I think the `n` argument should be provided only on the constructor call.
It would be something like:
```python
bearing = rs.BearingElement.from_excel(n=2, file='/path/to/bearing.xls')
```

flaviorangel, 2019-04-12:
Because python doesn't allow more than one constructor per class, I wrote a separated routine that returns a bearing object. You can check it [here](https://github.com/flaviorangel/ross/blob/inputs-from-table/ross/bearing_seal_element.py). I called it bearing_element_from_excel. Now, should blank cells be allowed? (since some arguments don't need to be provided)

I will now work on csv and toml conversion.

flaviorangel, 2019-04-12:
As for the toml, in the way we are saving bearing elements now, we only keep the first value of each array of arguments, except for  `w`. As in save_bearing_seal_element (rotor_assembly.py), we do:
`
data[file_name[:-6]][str(element.n)] = {
            "n": element.n,
            "kxx": element.kxx.coefficient[0],
            "cxx": element.cxx.coefficient[0],
            "kyy": element.kyy.coefficient[0],
            "kxy": element.kxy.coefficient[0],
            "kyx": element.kyx.coefficient[0],
            "cyy": element.cyy.coefficient[0],
            "cxy": element.cxy.coefficient[0],
            "cyx": element.cyx.coefficient[0],
            "w": w,
        }
`
Is that correct?

flaviorangel, 2019-04-25:
I wrote a static method to convert from table to toml, since there is no need to instantiate an object to do so. It returns a dictionary that can be saved as a toml file. This one can be loaded by ross later. I will create the PR and we can discuss there.

PR #95 (2019-04-28): Add methods to allow inputs from table in bearing element
Two methods were added to bearing element. The first one allows instantiating a bearing object from a table. The second one converts from table to dictionary, that can be saved as a toml file.
Close #79
closed 2019-04-28 by GabrielBachmann