petrobras/ross #132: Fluid flow is not included when installing via PIP

GabrielBachmann, 2019-05-30:


raphaeltimbo, 2019-05-30:
Just adding the traceback:
```python
---------------------------------------------------------------------------
ModuleNotFoundError                       Traceback (most recent call last)
<ipython-input-1-d5b423dbc569> in <module>
      1 import numpy as np
      2 import matplotlib.pyplot as plt
----> 3 import ross as rs

/srv/conda/envs/notebook/lib/python3.7/site-packages/ross/__init__.py in <module>
      1 __version__ = "0.0.1"
      2 
----> 3 from .bearing_seal_element import *
      4 from .disk_element import *
      5 from .materials import *

/srv/conda/envs/notebook/lib/python3.7/site-packages/ross/bearing_seal_element.py in <module>
     12 
     13 from ross.element import Element
---> 14 from ross.fluid_flow import fluid_flow as flow
     15 
     16 __all__ = ["BearingElement", "SealElement"]

ModuleNotFoundError: No module named 'ross.fluid_flow'
```

flaviorangel, 2019-05-31:
I was able to solve this bug by placing an empty "__init__.py" file in the fluid_flow folder. As explained in [setuptools documentation](https://setuptools.readthedocs.io/en/latest/setuptools.html#using-find-packages), find_packages will only recognized packages if they include an __init__.py file. I could not upload the same version to PyPI, so I changed it to 0.0.2.

closed 2019-05-31 by flaviorangel