petrobras/ross #144: Campbell with strange behavior for precession

raphaeltimbo, 2019-06-03:
For the rotor_example available in the rotor_assembly module the Campbell diagram is showing a strange behavior for one of the modes, with the precession changing from forward to backward:
[image]

We should investigate why this is happening. Values for kappa in this example are very low and can change depending on the environment in which we are running the code (travis ci, for example, was returning different values from those that I have obtained locally).

rodrigomoliveira1, 2019-06-06:
I guess I found the problem. When `whirl_to_cmap` returns a value for representing backward, forward or mixed precession, it is appended to an array called `results` in `def run_campbell`. So, when `whirl_to_cmap` returns 0 (forward) or 1 (backward), all the terms inside this array automatically become an "int". So, every time `whirl_to_cmap` would return a 0.5 (mixed) it was being turned into a "int", that is, the 0.5 would become 0.

rodrigomoliveira1, 2019-06-06:
PR #152 should fix it.

rodrigomoliveira1, 2019-06-10:
PR #152 closes this issue

closed 2019-06-10 by rodrigomoliveira1