petrobras/ross #173: Improve plot_rotor (bokeh output)

GabrielBachmann, 2019-06-16:
The Rotor object's plot_rotor() method is not giving a good bokeh output, as we can see in the following picture. 

[image]

This is the rotor is from this [example](https://github.com/ross-rotordynamics/ross/blob/master/docs/examples/example_05_09_01.ipynb).

rodrigomoliveira1, 2019-06-18:
It happens when building rotors with few elements (from 2 to 4 elements in total). Because the graphic representation of disks and bearings are calculated according to the shaft element lenght, I suggest doubling the number of elements in this example. It should look better.

raphaeltimbo, 2019-06-28:
I don't know if the number of elements should have this influence on the rotor plot aesthetics. Maybe we should evaluate if we could calculate bearings/disks proportions based on other things. I don’t remember how the code for this calculation is implemented, maybe it would be too much effort to change the code for a case that will not be happening with ‘real rotors’. We should start making examples with some different rotors and see if this will still be an issue.

rodrigomoliveira1, 2019-07-01:
The number of elements have influence since the size of disks' marks are calculated base on the length of shaft elements (1/8 of the element size, to be more exact).
Maybe i can illustrate the problem with some images.

This one uses 5 elements (1 shaft element per section). The circle seens large
[image]

This one uses 10 elements (2 shaft element per section).
[image]

And this one uses 15 elements (3 shaft element per section).
[image]

As we can see, when more elements were added, the proportion to the image got better.
Notice that neither the rotor length does not change, nor the diameters, nor the zoom (in or out).

We can still use the same rotor models for examples. We only need to add a few more elements to get a better aesthetic.

raphaeltimbo, 2019-07-30:
I have just plotted the rotor of a compressor (disks not included) and I have some comments:
- Almost every element has a slenderness ratio < 1.6. I think we should separate this slenderness ratio analysis from the rotor_plot. We could create a `.check_slenderness_ratio()` method that will return the colored plot and the hover tool could be used to show the slenderness ratio for each element.
[image]

- Bearings are too small. As discussed before, I think we should try to make them proportional to the diameter of the element in which the bearing is inserted.
[image]

rodrigomoliveira1, 2019-07-30:
I agree with both your statements. I'll try to fix these issues soon, then.

rodrigomoliveira1, 2019-08-02:
I did some tests here and came to this model:

[image]

The proportions are based on the diamater of the element now.

raphaeltimbo, 2019-08-04:
Here it looks like it is getting the wrong diameter:
[image]

Notice that we have two shaft elements in the same node here, so we should get the maximum outer diameter to position the bearings.

rodrigomoliveira1, 2019-08-05:
Yes, there is this issue and some other modifications I need to do. But I'll do it in another PR, the changes would conflict with PR #232.

rodrigomoliveira1, 2019-08-07:
In the PR #232 about the `check_slenderness_ratio` method, I kept the attribute "check_sld" because I needed something to diferentiate the HoverTool selection. One Tool is called when using `plot_rotor` and the other one is called when using `check_slenderness_ratio`. But I did remove the SR attribute from the other method by the way.

I couldn't rewrite the HoverTool by calling the bokeh axis (bk_ax).

rodrigomoliveira1, 2019-08-09:
PR #232 has been merged and closes this issue

closed 2019-08-09 by rodrigomoliveira1