petrobras/ross #179: Include bokeh plots on documentation

GabrielBachmann, 2019-06-20:
I've found a way to include bokeh plots on sphinx automatically, there's a sphinx extension called nbsphinx that does this.

raphaeltimbo, 2019-06-20:
Great! I think it would be a good idea to first run a test with one of the example notebooks. This way we can check if the pros and cons of this tool.

GabrielBachmann, 2019-07-14:
I've tested the [nbsphinx](https://github.com/spatialaudio/nbsphinx) extension and the outputs can be seen as follows
[image]
[image]

raphaeltimbo, 2019-07-22:
@GabrielBachmann , can you open a pull request for this?

GabrielBachmann, 2019-07-23:
PR #215 closes this issue.

closed 2019-07-23 by GabrielBachmann