petrobras/ross #187: Binder not working

raphaeltimbo, 2019-07-02:
Binder is not working because we still have some use of pytest in the code:

```python
---------------------------------------------------------------------------
ModuleNotFoundError                       Traceback (most recent call last)
<ipython-input-1-5fb570945225> in <module>
----> 1 import ross as rs
      2 from bokeh.io import output_notebook
      3 import numpy as np
      4 import matplotlib.pyplot as plt
      5 

/srv/conda/envs/notebook/lib/python3.7/site-packages/ross/__init__.py in <module>
      1 __version__ = "0.1.1"
      2 
----> 3 from .bearing_seal_element import *
      4 from .disk_element import *
      5 from .materials import *

/srv/conda/envs/notebook/lib/python3.7/site-packages/ross/bearing_seal_element.py in <module>
      7 import numpy as np
      8 import pandas as pd
----> 9 import pytest
     10 import scipy.interpolate as interpolate
     11 import xlrd

ModuleNotFoundError: No module named 'pytest'
```

We should remove pytest from these parts since this library should only be needed for development.

GabrielBachmann, 2019-07-14:
PR  #200 closes this issue

raphaeltimbo, 2019-07-15:
We need to remove pytest from DiskElement as well.

GabrielBachmann, 2019-07-23:
PR #213 closes this issue.

closed 2019-07-23 by GabrielBachmann