petrobras/ross #195: Create new file - report_API

rodrigomoliveira1, 2019-07-09:
The objective here is to create a file which receives a rotor as input and run all the required analysis from the Standards. 

The initial ideas for inputing a rotor were:
- input a single rotor for single Standard analysis.
- load previous rotors from toml to run the analysis.

We can propose methods in this thread. I think we could start using these 2 main methods: 
- stability_lvl1: this function will list all the initial steps the Standard requires
- stability_lvl2: this function will list all the following steps the Standard requires, if the rotor fails to meet the requirements of stability_lvl1.

I will open a new branch to upload this file, so we can work together on it.

raphaeltimbo, 2019-07-11:
I will try to put here the methods and an initial docstring describing the expected behavior:

```python
class Report:
    def __init__(self, rotor):
        """Report according to standard analysis.

        Parameters
        ----------
        rotor : ross.Rotor

        """
        pass

    def unbalance_response(self, clearances):
        """Evaluates the unbalance response for the rotor.

        This analysis takes the critical speeds of interest, calculates the
        position and weight of the required unbalance and performs the analysis
        including:
         - Check if vibration at MCS is below the limit with the applied weight;
         - Check if the clearances are ok if the vibration deteriorate to the
         limit level;

        Parameters
        ----------
        clearances : dict
            Dict mapping between node and its clearance in meters.
            (e.g. clearances = dict(3=0.001, 5=0.002...)
        """

    def stability_level_1(self):
        """Stability analysis level 1.

        This analysis consider a anticipated cross coupling QA based on
        conditions at the normal operating point and the cross-coupling
        required to produce a zero log decrement, Q0.

        Components such as seals and impellers are not considered in this
        analysis.

        Parameters
        ----------
        (Check what we need to calculate the applied cross coupling and list
         them as parameters)
        """
        pass

    def stability_level_1(self):
        """Stability analysis level 2.

        For the level 2 stability analysis additional sources that contribute
        to the rotor stability shall be considered such as:
        a)  labyrinth seals;
        b)  damper seals;
        c)  impeller/blade flow aerodynamic effects;
        d)  internal friction.

        Parameters
        ----------
        (Check what we need to calculate the applied cross coupling and list
        them as parameters)
        """
```

rodrigomoliveira1, 2019-07-11:
Nice, @raphaeltimbo!
Right now I'm working on an way to plot this graph below, when performing a frequency/unbalance response analysis:

[image]

I think it would be a good first output for the user to receive

rodrigomoliveira1, 2019-07-12:
@raphaeltimbo, I have just opened a new branch and created a new file called "api_report" (we can change the name later). I copied the concept you commented here, just to get a head start. 

I have also modified results.py and rotor_assembly.py. Some methods will have "report=False". this attribute will be True only when running the API file. The frequency response will look like that:

[image]

raphaeltimbo, 2019-07-12:
> @raphaeltimbo, I have just opened a new branch and created a new file called "api_report" (we can change the name later). I copied the concept you commented here, just to get a head start.
>
> I have [...]

It is looking really good!
Just one comment, I don't think that we should alter results.py and rotor_assembly.py by adding 'report=False'. The api_report should be decoupled from these modules. 
What I mean is that the results obtained from the run_methods should only be consumed by the api_report.
Notice that everything we need to generate the api analysis should be returned by each run method.
If we need to modify anything (e.g. add separation margin to the figure), this should be done with the data/figure that is returned, otherwise, we risk making our results methods too complicated, with several if blocks and so on.

rodrigomoliveira1, 2019-07-12:
I see. So, I'll think of some different coding structure. Maybe if I make more use of the **kwarg tool, the api file can add data to the plots generated by results.py.
I'm also trying to make a widget with bokeh to plot a table with some useful output data about the API analysis. When I have something done, I'll post here.

rodrigomoliveira1, 2019-07-15:
I added a table to the plot. I accept some ideas of relevant data we could organize on it.
Look:

[image]

rodrigomoliveira1, 2019-07-23:
This issue is being separated in smaller issues:
- Issue #210 
- Issue #211 

So, since the file has been already created, I'm closing this issue.

closed 2019-07-23 by rodrigomoliveira1