petrobras/ross #196: Change shaft .from_table() implementation

raphaeltimbo, 2019-07-09:
I have added some tests to the `ShaftElement.from_table()` method (see 895d57927d0775243712f27b6fca7aeba0636b42). Currently, they are marked with `@pytest.mark.skip()` to avoid failures on our CI.
The `.from_table()` method needs to be modified to consider the following:
 - User can use a table with SI or US units. When calling the method, the user should not have to specify which units he is using, this can be extracted from the spreadsheet (cells with 'meters' for example show that we are using SI);
- The method should have a default argument `sheet="Model"`, that points to the sheet inside the spreadsheet that has the model information;

@flaviorangel can you take a look at this? Tables used for the tests are at `/ross/tests/data/`.
After modifying the method we should remove the `@pytest.mark.skip()` from the test.

flaviorangel, 2019-07-10:
Sure. I'll take a look

flaviorangel, 2019-07-11:
I'm not sure if I'm doing it right, but I'm assuming the parameter "n" of the shaft as column elemnum -1.
I was able to write a routine that is working good for the metric system, except for the last assertion in the test: `assert_allclose(shaft[9].material.E, 6894.75)`. None of the provided materials have such value for the Elastic Modulus E. Maybe that was supposed to test for 1.0 instead of 6894.75?

flaviorangel, 2019-07-11:
> I'm not sure if I'm doing it right, but I'm assuming the parameter "n" of the shaft as column elemnum -1.
> I was able to write a routine that is working good for the metric system, except for the [...]

I noticed the same value as 1.0 in the US sheet. When multiplied by the conversion factor of 6894.757, it matches the assertion. Since both sheets are supposed to return the same shaft elements, one of the two must be wrong.

raphaeltimbo, 2019-07-11:
> I noticed the same value as 1.0 in the US sheet. When multiplied by the conversion factor of 6894.757, it matches the assertion. Since both sheets are supposed to return the same shaft elements, one [...]

You are right. Let's change the 1.0 from the SI sheet to 6894.75.

PR #199 (2019-07-12): Add model sheet instantiation for shaft element
The user can now instantiate shaft elements using a model sheet.
Close #196
closed 2019-07-12 by raphaeltimbo