petrobras/ross #202: Add medium size bearing to fluid flow

flaviorangel, 2019-07-14:
This issue came from issue #37 being broken into different smaller issues. In order to close it, we need to add a test example (preferably with data collected from a real situation) showing that the results match the expected. Since the pressure in a medium size bearing cannot be analytically calculated, it would be better to have real data to comparece with. If not available, we must think of how to validate the experiment.

raphaeltimbo, 2019-11-26:
@flaviorangel @JuliaMota 
Do we still have work to do on this issue?

JuliaMota, 2019-12-01:
We can already calculate pressure, forces and stiffness coefficients for medium bearings (only the damping coefficients are missing as mentioned in issue #261 ). However, we have not yet found a reference that provides input data and results for comparison in the case of the medium bearing.

raphaeltimbo, 2020-01-21:
We don't have a reference case for medium size, but short and long bearings were solved.

closed 2020-01-21 by raphaeltimbo