petrobras/ross #214: Bug in plot_eccentricity() method

GabrielBachmann, 2019-07-23:
I was changing the plots from example_05_05_01 (example 9) to the bokeh version of them.
With the following lines:
`# Plotting the eccentricity `
`my_fluid_flow.plot_eccentricity()`
I've got the following error log:
[image]

flaviorangel, 2019-07-24:
I'll check it.

flaviorangel, 2019-07-24:
I was not able to reproduce the error. I'm getting the plot as it was supposed to happen. This error seems to be related to bokeh. Are you on a different version?

[image]

GabrielBachmann, 2019-07-25:
I've tried updating my bokeh version, and I'm getting the same error...
The version that you are using of ross is the same as in pypi?

flaviorangel, 2019-07-25:
I was using the latest version on GitHub. How about you?
I updated ross on PyPI

flaviorangel, 2019-07-25:
I have tried updating bokeh and still could not reproduce the error. Does it happen to you outside jupyter notebook?

GabrielBachmann, 2019-07-27:
This issue was fixed with the PyPI update, it can be closed.

closed 2019-07-27 by GabrielBachmann