petrobras/ross #218: HoverTool does not work with shaft tapered element

rodrigomoliveira1, 2019-07-26:
HoverTool from Bokeh does not currently work with the glyph used to draw the shaft tapered element (glyph "patch"), according to bokeh website:
[image]

So, the informations about the shaft tapered element will not show up. Only for the shaft element.
I don't know if leaving it this way for now or drawing this element in a different way ("patch" was the most suitable glyph to draw a trapeze).

raphaeltimbo, 2019-07-29:
What I can see as options to solve this problem:
- Using patches (https://stackoverflow.com/a/53384398) instead of patch.
- Adding a transparent glyph together with the patch so that this transparent glyph will have the information to be displayed by the hover tool.

rodrigomoliveira1, 2019-07-29:
I did with the first option. The ColumnDataSource becomes a list of a single list but it's ok.

rodrigomoliveira1, 2019-07-31:
PR #232 closes this issue

closed 2019-07-31 by rodrigomoliveira1