petrobras/ross #219: Plotting Shaft Tapered Elements

rodrigomoliveira1, 2019-07-26:
I built an example using tapered elements in the shaft ends. And I'd like your opinion about the aesthetics.
The main question here is about the connection between the bearing and the shaft's end. For now, I am positioning (about the vertical axis) according to the mean value of the diameters. Do you agree with it or should it be positioned with the lowest diameter?

[image]

raphaeltimbo, 2019-07-29:
I am ok with the solution that you provided.
But these cases make me think if we really should be using a square to represent bearings, or if we should simply use springs/dampers to do that.

rodrigomoliveira1, 2019-07-29:
I'm trying to draw a spring/damper system. I will plot a figure here when I am done, so we can compare the results.

atbrandao, 2019-08-08:
I believe it is not a big issue for tapered elements, because I can't think of a real rotor in which we have bearing connected to a tapered section of the rotor.

However, I do like the idea of representing the bearing as spring/damper icon better than the current 'square' solution.

rodrigomoliveira1, 2019-08-09:
Since the shaft tapered element plotting style is ok, I'm opening another Issue (#248) to discuss the bearing plotting style.

closed 2019-08-09 by rodrigomoliveira1