petrobras/ross #223: Publish joss article

raphaeltimbo, 2019-07-29:
We need to see their checklist before submitting:
[joss website](https://joss.theoj.org/)

flaviorangel, 2019-08-01:
I will list here the requirements and state of each one. They were copied from ["Submitting a paper" page](https://joss.readthedocs.io/en/latest/submitting.html) and ["Review criteria" page](https://joss.readthedocs.io/en/latest/review_criteria.html).

- Make software available in an open repository (GitHub, Bitbucket etc.).
- Include an OSI approved open source license.
- Make sure that the software complies with the JOSS review criteria (see below).
- Author a short Markdown paper with a title, summary, author names, affiliations, and key references. See an example [here](https://github.com/arfon/fidgit/tree/master/paper).
- (Optional) create a metadata file and include it in your repository describing your software. [This script](https://gist.github.com/arfon/478b2ed49e11f984d6fb) automates the generation of this metadata.

flaviorangel, 2019-08-01:
JOSS review criteria:

The JOSS paper (the compiled PDF associated with this submission) should only include:

- A list of the authors of the software.
- Author affiliations.
- A summary describing the high-level functionality and purpose of the software for a diverse, non-specialist audience.
- A clear statement of need that illustrates the purpose of the software.
- Mentions (if applicable) of any ongoing research projects using the software or recent scholarly publications enabled by it.
- A list of key references including a link to the software archive.

flaviorangel, 2019-08-01:
Review items:

- Software license
- Documentation
- Statement of need
- Installation instructions
- Example usage
- API documentation
- Community guidelines
- Functionality
- Tests

raphaeltimbo, 2019-08-02:
> Review items:
> 
> * [x]  Software license
>   Acceptable: A plain-text LICENSE file with the contents of an OSI approved license
>   Not acceptable: A phrase such as ‘MIT license’ in a README file
>   Ours: We meet this requirement with our [license file](https://github.com/ross-rotordynamics/ross/blob/master/LICENSE.md). MIT License is also listed in choosealicense.com, appointed as preferred source of licenses by Joss.
> * [ ]  Documentation
>   Joss requires that the documentation is sufficient for the reviewer to understand the core functionality of the software. I believe we meet this requirement, but they also require that a high-level overview of this documentation should be included in a README file or equivalent, containing: statement of need, installation instructions, and example usage. We are not meeting a few conditions in each one (see below).
> * [ ]  Statement of need
>   The authors should clearly state what problems the software is designed to solve and who the target audience is. I believe we are lacking a stronger statement of need, as our [current one](https://ross-rotordynamics.github.io/ross-website/) ("Ross is a library written in Python for rotordynamic analysis") does not mention in details what Ross is capable of nor the target audience.

We need to update our readme file. That will not be difficult since we already have some of the required information in other parts of the documentation.

> * [ ]  Installation instructions
>   As for installation instructions, they require a clearly-stated list of dependencies. Ideally these should be handled with an automated package management solution:
>   Good: A package management file such as a Gemfile or package.json or equivalent
>   OK: A list of dependencies to install
>   Bad (not acceptable): Reliance on other software not listed by the authors
>   I know we have a list of requirements in our [setup.py](https://github.com/ross-rotordynamics/ross/blob/master/setup.py) file, but does that count as a package management file?

I believe the setup.py file counts as a package management file. We could also take a look at joss's issue tracker to see if we can find some discussion about this.

> * [ ]  Example usage
>   The authors should include examples of how to use the software (ideally to solve real-world analysis problems). We have examples in our [example page](https://ross-rotordynamics.github.io/ross-website/examples.html), but do we lack real-world analysis problems? Do they mean we should present real data?

I think our examples are ok, but we could also include some examples with real equipment. I believe we are going to have that soon.

> * [ ]  API documentation
>   Reviewers should check that the software API is documented to a suitable level.
>   Good: All functions/methods are documented including example inputs and outputs
>   OK: Core API functionality is documented
>   Bad (not acceptable): API is undocumented
>   Currently we classify as OK, since not all function and methods are documented with examples.

I also think we are ok, but we need to organize how we show that on the website.

> * [x]  Community guidelines
>   There should be clear guidelines for third-parties wishing to:
>   -Contribute to the software
>   -Report issues or problems with the software
>   -Seek support
>   We have all of these in our [Contributing page](https://ross-rotordynamics.github.io/ross-website/contributing.html).
> * [ ]  Functionality
>   Reviewers are expected to install the software they are reviewing and to verify the core functionality of the software. We still need beta testers to make sure someone outside the project can follow our instructions.

Agree. I will try to have some colleagues testing it more in the following weeks.

> * [x]  Tests
>   Authors are strongly encouraged to include an automated test suite covering the core functionality of their software.
>   Good: An automated test suite hooked up to an external service such as Travis-CI or similar
>   OK: Documented manual steps that can be followed to objectively check the expected functionality of the software (e.g. a sample input file to assert behaviour)
>   Bad (not acceptable): No way for you the reviewer to objectively assess whether the software works

raphaeltimbo, 2019-10-14:
We need to have the [orcid](https://orcid.org/) for each author.
@flaviorangel @GabrielBachmann @thiagoritto @JuliaMota @juvianna @rodrigomoliveira1

flaviorangel, 2019-10-15:
We can get a preview of the paper here:
https://whedon.theoj.org/

Just type 'https://github.com/ross-rotordynamics/ross' and hit 'compile'. The references are not showing up though. I'm investigating what is wrong.

flaviorangel, 2019-11-05:
> The references are not showing up though. I'm investigating what is wrong.

The references were not showing because they were not being mentioned in the paper. I thought it was not necessary to mention them, but I just checked a few other papers and they do mention the references that show up. Also, I tested our bib file by creating another repository and citing all our references. It compiles, as can be seen in the attached pdf, thus there is nothing wrong with them.

So now we must work on improving our paper with a few citations.
 
[f1rnvq5rgv5xw3pe8953.pdf](https://github.com/ross-rotordynamics/ross/files/3810236/f1rnvq5rgv5xw3pe8953.pdf)

raphaeltimbo, 2019-11-26:
I believe we have almost everything to proceed with the submission.
The only thing missing is to have tutorial and examples working.
This is being discussed on issue #381.

raphaeltimbo, 2019-12-02:
I've submitted the paper. Now we have to wait for the review process.
@thiagoritto @juvianna @rodrigomoliveira1 @GabrielBachmann @flaviorangel @JuliaMota 

[image]

rodrigomoliveira1, 2019-12-05:
To track our subscription to JOSS (pre-review), click at this [link](https://github.com/openjournals/joss-reviews/issues/1929).

raphaeltimbo, 2020-01-21:
I think we can close this now since we have already submitted the article.

closed 2020-01-21 by raphaeltimbo