petrobras/ross #226: Add .from_table method to disks

raphaeltimbo, 2019-07-29:
We need to implement the .from_table() method for the disks.
For now, I have submitted a PR #225 to add a small test and to modify the shaft spreadsheet so that they include a 'More' sheet with the disks information.

flaviorangel, 2019-07-30:
Column names in the "More" sheet differs a little from our argument names. Whereas we expect 'Id' and 'Ip', the file has 'Ip' and 'It'. I'm assuming 'It' and 'Id' to be the same and will prepare the method to expect one or another.

raphaeltimbo, 2019-07-30:
I confirm that 'It' and 'Id' are the same.

flaviorangel, 2019-07-30:
I had to modify the test parameters a bit. They seemed to be related to a different disk (disk 1 instead of 0, since in disk 0 both Ip and It were 0.0 and the mass was greater).

PR #231 (2019-08-06): Add from_table class method to disk element and general from_table method
Close #226 and close #228
closed 2019-08-06 by raphaeltimbo