petrobras/ross #236: Improve documentation

flaviorangel, 2019-08-06:
This issue is related to issue #223. In order to publish in Joss, we need to improve our documentation. I'll be listing here the functions/methods that need their documentation to be improved (and why).

flaviorangel, 2019-08-06:
In rotor_assembly.py:

- [ ] eq (lacks documentation);
- [ ] w (lacks documentation);
- [ ] run_modal (lacks documentation);
- [ ] whirl_direction (lacks output and example);
- [ ] whirl_values (lacks output and example);
- [ ] orbit (lacks documentation);
- [ ] _lti (lacks output and example);
- [ ] transfer_matrix (lacks documentation);
- [ ] run_freq_response (lacks example);
- [ ] run_forced_response (lacks documentation);
- [ ] _unbalance_force (lacks input, output, and example);
- [ ] unbalance_response (lacks example);
- [ ] time_response (lacks example);
- [ ] _plot_rotor_matplotlib (lacks documentation);
- [ ] _plot_rotor_bokeh (lacks example);
- [ ] run_mode_shapes (lacks documentation);
- [ ] plot_ucs (lacks example);
- [ ] plot_level1 (lacks example);
- [ ] run_time_response (lacks example);
- [ ] save_mat (lacks input and example);
- [ ] save (lacks example);
- [ ] load (lacks example);
- [ ] available_rotors (lacks documentation);
- [ ] remove (lacks documentation);
- [ ] run_static (lacks example);
- [ ] from_section (lacks output);
- [ ] rotor_regions (lacks documentation);
- [ ] MAC (lacks input, output, and example);
- [ ] MAC_modes (lacks input, output, and example);
- [ ] whirl (lacks input, output, and example);
- [ ] whirl_to_cmap (lacks input, output, and example).

flaviorangel, 2019-08-06:
In bearing_seal_element.py:

- [ ] class _Coefficient (lacks documentation, along with its methods)
- [ ] class _Stiffness_Coefficient (lacks documentation, along with its plot method)
- [ ] class _Damping_Coefficient (lacks documentation, along with its plot method)

From BearingElement class:
- [ ] repr (lacks documentation)
- [ ] eq (lacks documentation)
- [ ] isinstance (lacks documentation)
- [ ] save (lacks documentation)
- [ ] load (lacks documentation)
- [ ] dof_mapping (lacks documentation)
- [ ] M (lacks documentation)
- [ ] K (lacks documentation)
- [ ] C (lacks documentation)
- [ ] G (lacks documentation)
- [ ] patch (lacks example)
- [ ] bokeh_patch (lacks example)
- [ ] table_to_toml (lacks example)
- [ ] from_table (lacks example)
- [ ] from_fluid_flow (lacks example)

From class SealElement
- [ ] class SealElement (lacks documentation)
- [ ] patch (lacks example; also, arguments don't match documentation input)
- [ ] bokeh_patch (lacks example; also, arguments don't match documentation input)

flaviorangel, 2019-08-06:
In disk_element.py:

- [ ] eq (lacks documentation)
- [ ] repr (lacks documentation)
- [ ] save (lacks documentation)
- [ ] load (lacks documentation)
- [ ] dof_mapping (lacks documentation)
- [ ] K (lacks documentation)
- [ ] C (lacks documentation)
- [ ] patch (lacks example; also, arguments don't match documentation input)
- [ ] bokeh_patch (lacks example; also, arguments don't match documentation input)
- [ ] from_table (lacks example)

flaviorangel, 2019-08-06:
In element.py:

- [ ] save (lacks documentation)
- [ ] load (lacks documentation)
- [ ] C (lacks inputs)
- [ ] K (lacks inputs)
- [ ] load_data (lacks documentation)
- [ ] dump_data (lacks documentation)

flaviorangel, 2019-08-06:
In materials.py:

- [ ] eq (lacks documentation)
- [ ] repr (lacks documentation)
- [ ] str (lacks documentation)
- [ ] dump_data (lacks documentation)
- [ ] load_data (lacks documentation)
- [ ] use_material (lacks documentation)
- [ ] remove_material (lacks documentation)
- [ ] available_materials (lacks documentation)
- [ ] save_material (lacks documentation)

flaviorangel, 2019-08-07:
In results.py:

From class Results:

- [ ] __array_finalize__ (lacks documentation)
- [ ] __reduce__ (lacks documentation)
- [ ] __setstate__ (lacks documentation)
- [ ] save (lacks documentation)
- [ ] plot (lacks documentation)

From class CampbellResults:

- [ ] class CampbellResults (lacks documentation)
- [ ] _plot_matplotlib (lacks documentation)
- [ ] _plot_bokeh (lacks documentation)
- [ ] plot (lacks example; also, arguments don't match documentation input)

From class FrequencyResponseResults

- [ ] class FrequencyResponseResults (lacks documentation)
- [ ] plot_magnitude_matplotlib (lacks example; also, arguments don't match documentation input)
- [ ] plot_magnitude_bokeh (lacks example; also, arguments don't match documentation input)
- [ ] plot_phase_matplotlib (lacks example)
- [ ] plot_phase_bokeh (lacks example; also, arguments don't match documentation input)
- [ ] _plot_matplotlib (lacks example)
- [ ] _plot_bokeh (lacks example)
- [ ] plot (lacks example; also, arguments don't match documentation input)
- [ ] plot_freq_response_grid (lacks example)

From class ForcedResponseResults

- [ ] class ForcedResponseResults (lacks documentation)
- [ ] plot_magnitude_matplotlib (lacks example)
- [ ] plot_magnitude_bokeh (lacks example)
- [ ] plot_phase_matplotlib (lacks example)
- [ ] plot_phase_bokeh (lacks example)
- [ ] _plot_matplotlib (lacks documentation)
- [ ] _plot_bokeh (lacks documentation)
- [ ] plot (lacks example)

From class ModeShapeResults:

- [ ] class ModeShapeResults (lacks documentation)
- [ ] plot (lacks documentation)

From class StaticResults:

- [ ] class StaticResults (lacks documentation)
- [ ] plot (lacks example)

From class ConvergenceResults:

- [ ] class ConvergenceResults (lacks documentation)
- [ ] plot (lacks example)

From class TimeResponseResults:

- [ ] class TimeResponseResults (lacks documentation)
- [ ] _plot_matplotlib (lacks documentation)
- [ ] _plot_bokeh (lacks documentation)
- [ ] plot (lacks example; also, arguments don't match documentation input)

flaviorangel, 2019-08-07:
In shaft_element.py:

From class ShaftElement:

- [ ] eq (lacks documentation)
- [ ] save (lacks documentation)
- [ ] load (lacks documentation)
- [ ] from_table (lacks example)
- [ ] n (lacks documentation)
- [ ] dof_mapping (lacks documentation)
- [ ] repr (lacks documentation)
- [ ] str (lacks documentation)
- [ ] patch (lacks example; also, arguments don't match documentation input)
- [ ] bokeh_patch (lacks example; also, arguments don't match documentation input)

From class ShaftTaperedElement:

- [ ] eq (lacks documentation)
- [ ] save (lacks documentation)
- [ ] load (lacks documentation)
- [ ] n (lacks documentation)
- [ ] dof_mapping (lacks documentation)
- [ ] repr (lacks documentation)
- [ ] str (lacks documentation)
- [ ] C (lacks example)
- [ ] patch (lacks example; also, arguments don't match documentation input)
- [ ] bokeh_patch (lacks example; also, arguments don't match documentation input)

flaviorangel, 2019-08-07:
In utils.py:

- [ ] read_table_file (lacks example)

flaviorangel, 2019-08-07:
In fluid_flow.py:

- [ ] calculate_pressure_matrix_analytical (lacks output and example)
- [ ] calculate_coefficients (lacks example)
- [ ] mounting_matrix (lacks example)
- [ ] resolves_matrix (lacks example)
- [ ] calculate_pressure_matrix_numerical (lacks output and example)
- [ ] internal_radius_function (lacks example)
- [ ] external_radius_function (lacks example)
- [ ] get_rotor_load (lacks example)
- [ ] modified_sommerfeld_number (lacks example)
- [ ] sommerfeld_number (lacks example)
- [ ] calculate_eccentricity_ratio (lacks example)
- [ ] get_analytical_stiffness_matrix (lacks example)
- [ ] get_analytical_damping_matrix (lacks example)
- [ ] plot_eccentricity (lacks example)
- [ ] plot_pressure_z (lacks example)
- [ ] plot_shape (lacks example)
- [ ] plot_pressure_theta (lacks example)
- [ ] matplot_eccentricity (lacks example)
- [ ] matplot_pressure_z (lacks example)
- [ ] matplot_shape (lacks example)
- [ ] matplot_pressure_theta_cylindrical (lacks example)
- [ ] matplot_pressure_theta (lacks example)

flaviorangel, 2019-09-11:
Because there are so many different files to work on, and more than one person working on them, we agreed to let the others know in which file we are working on. So I'm on bearing_seal_element.py

rodrigomoliveira1, 2019-09-11:
I'm working on rotor_assembly.py

rodrigomoliveira1, 2019-09-19:
I've been working on results.py documentation and I wanted to raise two issues:
- I did not improved anything in `class Results` because it's obsolete and it's going to be removed soon.
- I did not implemented any example because if we calls rotor_example() in results.py, we get a recursive error. We cannot import rotor_assembly.py in results.py if rotor_assembly.py already imports results.py

flaviorangel, 2019-09-20:
I'll be working on element.py. There is an open PR for materials.py, so we got it all covered.

flaviorangel, 2019-09-20:
> * I did not improved anything in `class Results` because it's obsolete and it's going to be removed soon.
Okay. Once it is removed, we no longer have to worry about it.

> * I did not implemented any example because if we calls rotor_example() in results.py, we get a recursive error. We cannot import rotor_assembly.py in results.py if rotor_assembly.py already imports results.py
This is a tricky one. I'm not sure how we should handle this. Maybe we could leave it without examples and see if the reviewers will require some.

raphaeltimbo, 2019-11-26:
I believe we can close this for now.

closed 2019-11-26 by raphaeltimbo