petrobras/ross #238: Question

juvianna, 2019-08-06:
There are two repositories in https://github.com/ross-rotordynamics. One called ross and another ross-website, where it is supposed to have a website, right? But I can only find the website at the bottom of the ross repository through a hyperlink "Access the documentation here". Is it correct?

flaviorangel, 2019-08-07:
The ross-website repository holds the HTML files, but it is not supposed to have a website. It just stores the files in a Git way. It was separated from the main repository because HTML files are usually larger and automatic generated files that mess with develop metrics, such as the number of lines added to the project. For example, the api.html alone has more than 2k lines.

raphaeltimbo, 2019-08-07:
As @flaviorangel pointed out, the ross-website repository holds the HTML files and the [GitHub Pages](https://pages.github.com/) service builds the website:
[image]

closed 2019-08-07 by raphaeltimbo