petrobras/ross #245: Include bearing housing dynamics

atbrandao, 2019-08-08:
It is important to be able to include in the modelling the stiffness and inertia of the bearing housing, external supports, squeeze filme dampers or any other kind of support dynamics other than the actual oil film.

We usually have mass and stiffness parameters of these supports. So a easy way to include this in the model is create a point mass element and connect it to the rotor via the oil film sitffness and to the ground via the support stiffness.

It would be best if the software allowed the user to create multiple point mass elements and connect them to any other two nodes, that might be a rotor node, another point mass node or the ground.
This way we can model the full system including, for example, the squeeze filme damper, the bearing housing and baseplate, when needed.

raphaeltimbo, 2019-09-04:
My ideas to implement this are:
- Create an additional `n_link` argument to bearing. We would use this argument to supply an additional node to which the bearing would connect. This would allow the user to connect bearing houses, squeeze film damper and co-axial rotors;
- Add additional routine to verify bearings that have `n_link != None` and change the global matrices accordingly;
- Add a routine to change the elements’ axial location to allow the correct drawing of the rotor. A possible solution here would be to add an attribute `axial_location` to the elements.

rodrigomoliveira1, 2019-09-04:
By the way, we could make It another class like SealElement is. Just to make sure what you are trying to represent.

atbrandao, 2019-09-05:
I think creating a new class, something like `PointMass`, is a good idea to organize this new feature.

This `PointMass` class would then be assigned a node number as an attribute, like `PointMass.n`. (obs.: This new element is 2-dimensional (X and Y), and the class should allow different mass values for X and Y, which can be useful in certain cases).

The sequence to use this to model support stiffness would go like this:

1. The first bearing element, representing the oil film, would receive `n = rotor_node` and `n_link = PointMass.n`.

2. And then a second bearing element, representing the supprt stiffness, would be created receiving only `n = PointMass.n` as argument. This would connect the point mass to ground with the stiffness of the support.

With this philosophy I think the code might be able to simulate things like co-axial rotors, as Timbó mentioned. However, the ideal way to do this would be with the creation of a different tool, like `RigidLink(n1 , n2)`, to rigidly link the nodes `n1` and `n2` in all 4 degrees of freedom, instead of flexibly coupling them using the `BearingElement`, which is 2-dimensional. I think this might be a matter for a different Issue.

raphaeltimbo, 2019-09-05:
> I think creating a new class, something like `PointMass`, is a good idea to organize this new feature.
>
> This `PointMass` class would then be assigned a node number as an attribute, like [...]

I like the idea of creating the `PointMass` class to implement this. I’ll try to do it that way.
Regarding the co-axial rotors, I believe we could have flexible or rigid links. Friswell’s example 6.6.1 links two rotors using a bearing. 
But I agree we should take care of this in another issue. I will open one for that.

PR #277 (2019-09-17): Add rotor link
The purpose of this PR is to add the possibility of simulating bearing housing and supports, as requested on #245.

It is still a working in progress.

- [x] Create class `PointMass`;
- [x] Add an additional argument `n_link` to bearing;
- [x] Change rotor assembly to check for links when assembling global matrices;
- [x] Add patches for drawing.

Closes #245.
closed 2019-09-17 by raphaeltimbo