petrobras/ross #246: Include new features in the tutorial Notebook

atbrandao, 2019-08-08:
This is both a suggestion and a request for help.

I'm having trouble learning how to use the code's new features, such as the inclusion of tapered elements.

I think it would be great if the tutorial Notebook was updated or a new example was created everytime a new feature is implemented.

raphaeltimbo, 2019-08-09:
Thanks for the suggestion @atbrandao !
We will try to add the new features to the documentation as soon as possible.
Regarding the tapered elements, we are now having some problems in the code regarding the stiffness matrix for these elements. I've just opened an issue on this: #247

rodrigomoliveira1, 2019-08-09:
In the last two days I did a test here in the lab with some people trying to install, learn and use ROSS.

I will list down here some feedback I received about the ROSS website:

- Installation tab:
    - Although we say that ROSS is tested in Python 3.6 or higher, we should clearly state that Anaconda download must be version 3.7.
    - Maybe add some prints of how to install Anaconda.

- Tutorial tab - Shaft_element
    - In the tutorial of how to instantiate a shaft element, we have this code here. They got a bit confused about the `l_list = [0.25 for _ in range(n)]`, saying it was not clear that `l_list` refers to the element length
    - Another source of misunderstanding is the use of the variable `n` to denote the number of elements to be created, since `n` is also indicated in the class explanation as the index of a given element. So it should be replace by another variable like `N` or something else.
    - The suggestion is changing the code to this:

```
L = 0.25
i_d = 0
o_d = 0.05
N = 6
l_list = [L for _ in range(N)]
shaft_elements = [
    rs.ShaftElement(
        L=l,
        i_d=i_d,
        o_d=o_d,
        material=steel,
        shear_effects=True,
        rotary_inertia=True,
        gyroscopic=True,
    )
    for l in l_list
]
```

- Tutorial tab - Load Materials
    - in input [5], we do `steel = rs.Material.use_material('Steel')`. We should standarize the object `steel` using capital letter to match the other code lines: `Steel = rs.Material.use_material('Steel')`

- Tutorial tab - Static Analysis
    - Add a new command line after input [26] (`static = rotor1.run_static()`) to explain how to return a plot:
```
[26]: static = rotor1.run_static()
[27]: static.plot()
```

raphaeltimbo, 2019-11-26:
Regarding the `ShaftTaperedElement`, we will treat this on issue #394.

closed 2019-11-26 by raphaeltimbo