petrobras/ross #310: Add patch for point mass element

raphaeltimbo, 2019-09-26:
We added a feature related to linking different supports (see #277). To accomplish this, it was necessary to add the PointMass element. Now we need to add a patch for this element so that it will be included on the rotor plot.
It may be necessary to change some parameters from the BearingElement patch so that the drawings don't overlap.

rodrigomoliveira1, 2019-09-29:
I've changed the patches from bearing and seal elements and point mass.
How it looks?

[image]

raphaeltimbo, 2019-09-30:
It looks good. I’ve already approved the PR.
Later, we could try to bring the point mass to the front of the plot, but this is just aesthetics.

closed 2019-09-30 by raphaeltimbo

rodrigomoliveira1, 2019-09-30:
It's already on the front. When I posted this image, the transparency was not at maximum, it was 0.8. But in the code I have set to 1.0. Besides, before commiting, I have moved the loop to plot the point mass elements to the end, so it appears on the front.

```
        # plot shaft elements
        for sh_elm in self.shaft_elements:
            position = self.nodes_pos[sh_elm.n]
            hover = sh_elm.bokeh_patch(position, check_sld, bk_ax)

        bk_ax.add_tools(hover)

        # plot disk elements
        for disk in self.disk_elements:
            position = (self.nodes_pos[disk.n], self.nodes_o_d[disk.n] / 2)
            hover = disk.bokeh_patch(position, bk_ax)

        bk_ax.add_tools(hover)

        mean_od = np.mean(self.nodes_o_d)
        # plot bearings
        for bearing in self.bearing_seal_elements:
            z_pos = self.df[self.df.tag == bearing.tag]["nodes_pos_l"].values[0]
            y_pos = self.df[self.df.tag == bearing.tag]["y_pos"].values[0]
            position = (z_pos, y_pos)
            bearing.bokeh_patch(position, mean_od, bk_ax)

        # plot point mass
        for p_mass in self.point_mass_elements:
            z_pos = self.df[self.df.tag == p_mass.tag]["nodes_pos_l"].values[0]
            y_pos = self.df[self.df.tag == p_mass.tag]["y_pos"].values[0]
            position = (z_pos, y_pos)
            hover = p_mass.bokeh_patch(position, bk_ax)

        bk_ax.add_tools(hover)

        return bk_ax
```