petrobras/ross #326: CAMPBELL - Plot the harmonics with different colors and legend

leoishimoto, 2019-10-02:
1. To provide differentiation in color and legend of the harmonics is desirable.
2. The legend of "Forward", "Mixed" and "Backward" modes should not have any color.
3. The red squares (intersections) should have a legend.
4. An argument to change units (rpm,  Hz), in all 3 axis (log dec or damping factor, Rotor speed and DNF's) is desirable.
5. The previous color map (from red to blue) was better for the visualization.

[image]

rodrigomoliveira1, 2019-10-03:
> To provide differentiation in color and legend of the harmonics is desirable.

Ok. Each harmonic will be ploted with a different color and will be displayed on the legend.

> The legend of "Forward", "Mixed" and "Backward" modes should not have any color.

I agree with you, however, bokeh does not offer an attribute in Legend to change the glyph color displayed at the board. The glyphs colors match the renderers. There are several options to change the legend colors after setting a render, but for text, background, border line, etc. Maybe, I find a way using bokeh's javascript.

> The red squares (intersections) should have a legend.

Sure, you are rigth, I'm adding it.

> An argument to change units (rpm, Hz), in all 3 axis (log dec or damping factor, Rotor speed and DNF's) is desirable.

Yes, this is something we need to apply to every other method we have. We'll working on it.

> The previous color map (from red to blue) was better for the visualization.

Bokeh does not offer a continuous palette (256 colors) of Red-to-Blue. Bokeh as a ColorBrewer palette called RdBu, but it has only 11 colors. Actually, I still can implement a 256 RdBu (as matplotlib has) using `colorcet` library, but it needs to be installed and imported.

The Figure below is an example using Bokeh's RdBu[11].
[image]

raphaeltimbo, 2019-10-04:
> Bokeh does not offer a continuous palette (256 colors) of Red-to-Blue. Bokeh as a ColorBrewer palette called RdBu, but it has only 11 colors. Actually, I still can implement a 256 RdBu (as [...]

It is possible to create a bokeh colormap from a matplotlib one as shown in [this SO answer](https://stackoverflow.com/a/49934218/5726899).

```python
from bokeh.colors import RGB
from matplotlib import cm

m_coolwarm_rgb = (255 * cm.coolwarm(range(256))).astype('int')
coolwarm_palette = [RGB(*tuple(rgb)).to_hex() for rgb in m_coolwarm_rgb]
```

rodrigomoliveira1, 2019-10-04:
> It is possible to create a bokeh colormap from a matplotlib one as shown in this SO answer.

Thanks, @raphaeltimbo! it worked just fine.
[image]

rodrigomoliveira1, 2019-10-04:
I found an way to display the glyphs with other color at the legend

[image]

rodrigomoliveira1, 2019-10-08:
PR #334 closes this issue. It can be re-opened later if something is missing.
The issue about setting units will evaluated later.

closed 2019-10-08 by rodrigomoliveira1