petrobras/ross #327: "run_static" - Documentation issues

leoishimoto, 2019-10-02:
The description in the documentation of the variables contained in the dictionary does not match with real ones. There are some mismatched capital letters. It would be nice to follow a standard.

[image]

Another suggestion is to include units in the dict.
Other possibility is to include all the data directly on the rotor object, without creating a dictionary. I did not understood the logic when an information became a dict or when is attribute to the rotor or to the results of the method.

rodrigomoliveira1, 2019-10-04:
`force_data` dictionary has been removed.
Now, the following items from `force_data` became Rotor's attributes:
- shaft_weight - Shaft total weight
- disk_forces - Weight forces of each disk
- bearing_reaction_forces - The static reaction forces on each bearing

`Rotor.run_static()` docstring's been updated

leoishimoto, 2019-10-04:
Just to clarify: Shaft total weight is the sum of the shaft elements or is the rotor (shaft + lumped mass)?
A summary of the rotor is a helpful tool to ensure that are no errors in the model, to perform double checks if the node is the rigth one and so on.

See the example from ROMAC:
[image]

They provide also the global nodes summary (there is a feature to create a automatic discretization of the rotor):

This is the model:
[image]

This is the visualization of the final nodes:
[image]

XLTRC2 also provides the summary:
[image]

Information as Inertias, CG location, Total length are all useful. I think you can store all of them as attributes, but maybe a method called "summary" that plots something as the two examples above may be a good idea.

rodrigomoliveira1, 2019-10-04:
> Just to clarify: Shaft total weight is the sum of the shaft elements or is the rotor (shaft + lumped mass)?
A summary of the rotor is a helpful tool to ensure that are no errors in the model, to perform double checks if the node is the rigth one and so on.

The attribute `shaft_total_weight` refers only to the shaft. Maybe it's better adding another attribute called `rotor_total_weight` considering the the shaft + lumped mass to avoid any missunderstanding about it.

> Information as Inertias, CG location, Total length are all useful. I think you can store all of them as attributes, but maybe a method called "summary" that plots something as the two examples above may be a good idea.

It sounds a good idea. For now we have the dataframes built using Pandas. But I think I can organize it using Bokeh widgets and plot like the examples you've said. Thanks for the suggestion.

Soon, I'll come with a preview for what this Bokeh widget will be and we can discuss more about it.

rodrigomoliveira1, 2019-10-08:
This is a preview of what the summary table will become when it's done.

[image]

It's quite simple to use it (**not implemented yet**):

```
>>> import ross as rs
>>> rotor = rs.rotor_example()
>>> table = rotor.summary().plot()
>>> show(table)
```

We can also create multiples types of summary and organize them in different plot() methods.

rodrigomoliveira1, 2019-10-11:
I did some improvements to this table. Now we can alternate through tabs to get access to different information. Take a look:

[image]

[image]

[image]

I've been looking for some way to customize colors, background, fontsize, etc, but to do this, we need to use JavaScript.

rodrigomoliveira1, 2019-10-15:
PR #342 closes this issue.

closed 2019-10-15 by rodrigomoliveira1