petrobras/ross #329: Method run_modal() improvements

leoishimoto, 2019-10-02:
It is necessary to specify the number of DOF that will be calculated in this method as an argument of the method.
For the same reason it is necessary to specify as an argument the speeds. Multiple speeds are desirable as bearings and seals coefficients change with speed.

For the plot is desirable to inform if it is a forward or backward mode (I assume blue is forward and red is backward but the legend should be available). In the title, the number of the mode should be informed (mode 1 - speed 1000 rpm).

The Bokeh option should be included as the manipulation of the plot is easier (zooming). Rotation of the mode with the mouse is important. Animation is desirable also.

[image]

raphaeltimbo, 2019-10-07:
> It is necessary to specify the number of DOF that will be calculated in this method as an argument of the method.

I will add an argument so that the user can say how many modes it should calculate.

> For the same reason it is necessary to specify as an argument the speeds. Multiple speeds are desirable as bearings and seals coefficients change with speed.

Here, the speed argument is only one speed and I think if the user wants more speeds to be calculated they can iterate on the speeds and store the results for each case.

> For the plot is desirable to inform if it is a forward or backward mode (I assume blue is forward and red is backward but the legend should be available). In the title, the number of the mode should be informed (mode 1 - speed 1000 rpm).

I will add these.

> The Bokeh option should be included as the manipulation of the plot is easier (zooming). Rotation of the mode with the mouse is important. Animation is desirable as well.

Last time we checked Bokeh did not have an option for 3d plots. We can still use it by wrapping a javascript library ([see example here](https://demo.bokeh.org/surface3d)). I believe this will take some work and currently it is not a priority, so I will open a separate issue for this one.
> 
> [image]

rodrigomoliveira1, 2019-11-12:
> For the plot is desirable to inform if it is a forward or backward mode (I assume blue is forward and red is backward but the legend should be available). In the title, the number of the mode should be informed (mode 1 - speed 1000 rpm).

PR #376 add this feature.

[image]

rodrigomoliveira1, 2019-11-13:
I think we can close this issue once PR #337 and #376 have been merged and we have all the requests implement except for plotting mode shapes using Bokeh (Issue #336).

closed 2019-11-13 by rodrigomoliveira1