petrobras/ross #332: Bearing is not connected to tapered element

raphaeltimbo, 2019-10-04:
In the rotor plot the bearing is not connected to tapered element:
[image]

I believe we should also have some scale factor for the bearings so we can increase/decrease bearing size on a case by case basis.

rodrigomoliveira1, 2019-10-04:
> In the rotor plot the bearing is not connected to tapered element:

Maybe if we implement a routine the checks the node diameters, we can choose where to fix the glyph, on the smaller diameter or on the largest diameter for a given node.

> I believe we should also have some scale factor for the bearings so we can increase/decrease bearing size on a case by case basis.

This is simpler. Here we can see where the bearings are called to be plotted
```
        mean_od = np.mean(self.nodes_o_d)

        # plot bearings
        for bearing in self.bearing_seal_elements:
            z_pos = self.df[self.df.tag == bearing.tag]["nodes_pos_l"].values[0]
            y_pos = self.df[self.df.tag == bearing.tag]["y_pos"].values[0]
            position = (z_pos, y_pos)
            bearing.bokeh_patch(position, mean_od, bk_ax)
```
A scale factor could be introduced by multiplying (or dividing) the `mean_od`. Every parameter inside the `BearingElement.patch()` is based on it. We could do something like:

```
        mean_od = np.mean(self.nodes_o_d) * scale_factor

        # plot bearings
        for bearing in self.bearing_seal_elements:
            z_pos = self.df[self.df.tag == bearing.tag]["nodes_pos_l"].values[0]
            y_pos = self.df[self.df.tag == bearing.tag]["y_pos"].values[0]
            position = (z_pos, y_pos)
            bearing.bokeh_patch(position, mean_od, bk_ax)
```

raphaeltimbo, 2019-10-07:
> This is simpler. Here we can see where the bearings are called to be plotted
>
> ```
>         mean_od = np.mean(self.nodes_o_d)
>
>         # plot bearings
>         for bearing in [...]

`y_pos = self.df[self.df.tag == bearing.tag]["y_pos"].values[0]` should already return the correct position where we would fix the bearing. I think there is some problem on the routine that fills the y_pos values when we instantiate the rotor.

> A scale factor could be introduced by multiplying (or dividing) the `mean_od`. Every parameter inside the `BearingElement.patch()` is based on it. We could do something like:
>
> ```
>         mean_od = [...]

This is a good approach, but I think it would be better that each bearing had its own scale factor so we can plot bearings and seals with different sizes for example. The scale factor could be an optional argument for the bearing/seal instantiation.

closed 2019-10-18 by rodrigomoliveira1