petrobras/ross #335: Error while generating the geometry plot of the rotor

rennergcm, 2019-10-04:
When I was trying to run the tutorial, the following error appeared. The probable cause might be my panda version (0.20.3) 
```python
TypeError                                 Traceback (most recent call last)
<ipython-input-47-19cb69e4ba3f> in <module>()
      2     shaft_elements,
      3     disks,
----> 4     bearings
      5 )

~\AppData\Local\Continuum\anaconda3\lib\site-packages\ross\rotor_assembly.py in __init__(self, shaft_elements, disk_elements, bearing_seal_elements, point_mass_elements, w, sparse, n_eigen, min_w, max_w, rated_w)
    278 
    279         df = pd.concat(
--> 280             [df_shaft, df_disks, df_bearings, df_point_mass, df_seals], sort=True
    281         )
    282         df = df.sort_values(by="n_l")

TypeError: concat() got an unexpected keyword argument 'sort'
```

rodrigomoliveira1, 2019-10-04:
> The probable cause might be my panda version (0.20.3)

This makes sense. The Pandas version 0.20.3 does not have the parameter `sort` for `pd.concat()`

To solve this, try updating Pandas:

1. Close all Anaconda applications
2. Open the Anaconda Prompt
3. Use the command `conda install -c anaconda pandas`. It will install the last version (0.25.1).
4. Try running rotor_assembly again

raphaeltimbo, 2019-10-07:
We should also adjust our setup to require pandas > 0.23.

PR #339 (2019-10-10): Add requirement for pandas >= 0.23
Closes #335.
commit d2469494 (2019-10-10): Add requirement for pandas >= 0.23

Closes #335.
closed 2019-10-10 by rodrigomoliveira1