petrobras/ross #336: Bokeh 3d mode shape plot

raphaeltimbo, 2019-10-07:
We are using matplotlib to plot mode shapes, since Bokeh does not have a native option for 3d plots. We can follow [this example](http://bokeh.pydata.org/en/latest/docs/user_guide/extensions_gallery/wrapping.html#userguide-extensions-examples-wrapping) to implement things in bokeh.

raphaeltimbo, 2020-01-21:
I will try to implement this one.

raphaeltimbo, 2020-02-10:
After checking for some options to implement this, I concluded that the option provided in the first post ([this example](http://bokeh.pydata.org/en/latest/docs/user_guide/extensions_gallery/wrapping.html#userguide-extensions-examples-wrapping)) is too complicated and I think there are better js libraries (visually) than the one used in the example (visjs).
Other option that I have been looking at is [plotly](https://plot.ly/python/).
When we started this project we actually discussed about using plotly or bokeh, but at the time plotly looked not so open as bokeh (it had an online and an offline version - [this is not the case anymore](https://medium.com/plotly/plotly-py-4-0-is-here-offline-only-express-first-displayable-anywhere-fc444e5659ee)).

After some tests I got something like this:
[image]

My conclusion is that plotly looks like a good option to implement this and it has nice 3d capabilities if we ever need.

Problems are: additional dependency (with ~4mb on download), and also slower import time for ross (but I have not measured that). Another point is that with this we would have 3 plotting libraris within ross (matplotlib, bokeh and plotly) which could be confusing to the user.

rodrigomoliveira1, 2020-02-11:
I've been looking for some reference about plotting 3D with bokeh too. However, reading their Issues, I don't see they are prioritizing the 3D library. Even the 2.0 milestone has issues related to that.

> My conclusion is that plotly looks like a good option to implement this and it has nice 3d capabilities if we ever need.
> Problems are: additional dependency (with ~4mb on download), and also slower [...]

I agree that adding another plotting library could get confusing for users.
Another option, I can think right now, is checking the plotly capabilities. If it covers what we need we could replace bokeh by plotly.

raphaeltimbo, 2020-02-12:
> I've been looking for some reference about plotting 3D with bokeh too. However, reading their Issues, I don't see they are prioritizing the 3D library. Even the 2.0 milestone hasn't issues related to that.
> 
> > My conclusion is that plotly looks like a good option to implement this and it has nice 3d capabilities if we ever need.
> > Problems are: additional dependency (with ~4mb on download), and also slower import time for ross (but I have not measured that). Another point is that with this we would have 3 plotting libraris within ross (matplotlib, bokeh and plotly) which could be confusing to the user.
> 
> I agree that adding another plotting library could get confusing for users.
> Another option, I can think right now, is checking the plotly capabilities. If it covers what we need we could replace bokeh by plotly.

I would be ok with changing to plotly. I believe it has the same capabilities that bokeh has (+ 3d).
The only problem is that they have a different api, so we would need to check how much work we would need to change all plots that we have now.

raphaeltimbo, 2020-02-21:
We have decided to test plotly in some of our functions. So, for a moment, we will have 3 different plot packages as dependencies. Later we will decide which library to keep. 
I will finish the 3d mode shape implementation and submit a PR.
Later I will open issues for different plots that we should try to implement using plotly.

rodrigomoliveira1, 2020-05-12:
I've trying Plotly this week, and it seems to cover all of our needs. However Plotly has a current Issue with its legend. 

You see, here in this example using stochastic ross to plot the Campbell Diagram, the legend box for both graphs are merged, and it gets a little confusing. Plotly does not offer a command option to separate legend acconding to the subplots, it merges them automatically.

A solution could be adding an argument called `legendgroup`, and a command to add a larger gap between the "grouped labels". But adding `legendgroup` makes the "hide scatter event" to hide / show all the scatter with the same "legendgoup" name, which means, if you click on any label at the legend box, all the scatters/lines disappear (we'd expect only the respective scatter/line to disappear)

[image]

rodrigomoliveira1, 2020-05-12:
I did some kind of bypass to this issue.

```text
    def plot(self, percentile=[], conf_interval=[], *args, **kwargs):
        fig0 = self.plot_nat_freq(
            percentile=percentile, conf_interval=conf_interval, *args, **kwargs
        )

        default_values = dict(showlegend=False)
        for k, v in default_values.items():
            kwargs.setdefault(k, v)

        fig1 = self.plot_log_dec(
            percentile=percentile, conf_interval=conf_interval, *args, **kwargs
        )
        subplots = make_subplots(rows=1, cols=2)

        for data in fig0["data"]:
            subplots.append_trace(data, 1, 1)
        for data in fig1["data"]:
            subplots.append_trace(data, 1, 2)

        subplots.update_xaxes(fig0.layout.xaxis, row=1, col=1)
        subplots.update_yaxes(fig1.layout.yaxis, row=1, col=1)
        subplots.update_xaxes(fig0.layout.xaxis, row=1, col=2)
        subplots.update_yaxes(fig1.layout.yaxis, row=1, col=2)
        subplots.update_layout(fig0.layout)

        return subplots
```

I set a default kwarg in `.plot()` to turn the legends off, only on the second plot. So, if you plot a graph individually, you still have the legends on. If you used `.plot()` to display both, you are returned this image:

[image]

And if you click on a legend label, you hide line in both plots:

[image]

rodrigomoliveira1, 2020-05-13:
@raphaeltimbo, I have replaced all the bokeh library from `ross.stochastic` with plotly. I try to replace all the ROSS files by tomorrow, and then, you can try to run ROSS importing from my fork to check the Plotly features

rodrigomoliveira1, 2020-05-15:
Look how Campbell Diagram looks with Plotly.

It took a little more work time than I thought it'd take. Although Plotly needs to receive more arguments than Bokeh, it seems to be simpler to customize the plots, and the code looks clearer and more organized than before.

The dictionary structure it uses is more objective

[image]

raphaeltimbo, 2020-05-18:
It is looking nice, @rodrigomoliveira1 ! The legend outside the plot is also much better (not going in front of the curves).

closed 2020-05-27 by raphaeltimbo