petrobras/ross #341: Folder "rotors" does not appear when installing ROSS

rodrigomoliveira1, 2019-10-11:
Yesterday I was testing the code during a class and when people installed ROSS, the folder "rotors" (which keeps the saved rotor models) was not being installed. Everything was running fine, but we were unable to save the files, only with we had created the folder manually.

rodrigomoliveira1, 2019-11-21:
I think we can solve this problem adding /rotors diretory to Manifest.in, as it's explained in this [PyPI tutorial](https://packaging.python.org/guides/using-manifest-in/)

raphaeltimbo, 2020-01-21:
@rodrigomoliveira1 will try to solve this using the MANIFEST.in file.

GabrielBachmann, 2020-01-21:
What about giving the option for the user to give a path setting the default value to the rotors folder, then checking if the folder exists(creating a folder if not), as the code bellow:

```python
def save(self, file_name, path_rotors=Path(os.path.dirname(ross.__file__))/'rotors'):
        if not path_rotors.is_dir() and path_rotors.suffix == '':
            path_rotors.mkdir()
```

raphaeltimbo, 2020-01-21:
@GabrielBachmann , we wanted to have the same behavior that we have in other Python libraries when we try to save a file.
For example, in pandas:
```python
>>> import pandas as pd
>>> df = pd.DataFrame({’name’: [‘Raphael’, ‘Donatello’],
... ‘mask’: [‘red’, ‘purple’],
... ‘weapon’: [‘sai’, ‘bo staff’]})
```
The following will save a `a.csv` file to the current working directory.
```python
>>> df.to_csv(path_or_buf=’a.csv’)
```
The following will raise an exception, since the `test/` directory does not exist:
```python
>>> df.to_csv(path_or_buf=’test/a.csv’)
FileNotFoundError: [Errno 2] No such file or directory: ‘test/a.csv’
```

So, if the user wants to save to a folder he needs to declare that in the ‘path’ argument. Having a default folder different from the current working directory is not the expected behavior for a Python library.

rodrigomoliveira1, 2020-02-21:
We changed how we save rotors model. The user now inputs where it should be saved.
The folder "rotors" is not necessary anymore.

closed 2020-02-21 by rodrigomoliveira1