petrobras/ross #367: Error in Campbell Diagram

raphaeltimbo, 2019-11-04:
The Campbell diagram for a specific compressor is showing mixed precession for some modes:
[image]

This is not the correct result.

rodrigomoliveira1, 2019-11-04:
Try instantiating `Rotor` setting `sparse=False` and check the result again. Maybe it's something about the kappa values when using `scipy.eigs()`. It's only a test to filter out what might be going wrong.
In any case we need to investigate what's causing this problem.

raphaeltimbo, 2019-11-04:
> Try instantiating `Rotor` setting `sparse=False` and check the result again. Maybe it's something about the kappa values when using `scipy.eigs()`. It's only a test to filter out what might be going wrong.
> In any case we need to investigate what's causing this problem.

Good suggestion. I've tried to check that and the results are the same.
I have noticed some inconsistency between the way the models are being imported from excel. I will double check the model before investigating anything on the Campbell routines.

rodrigomoliveira1, 2019-11-05:
Well, between having bugs with the campbell diagram and importing from excel, solving the import issue is much simpler.
If you like, send me the example you are trying to run and I can check it out too.

raphaeltimbo, 2019-11-12:
I confirm that this was an import issue and #374 solves this.

PR #374 (2019-11-13): Change imported index
As per #368, we had an inconsistency between imported shaft and disks.
PR #372 changed the shaft index, removing the lines that would change the index from 1 to 0 based. The problem is that the imported shaft and disks would be 1 indexed, and python (and ross) are 0 based, so I believe it is better to apply the 0 index to both, shaft and disks imported.

Closes #367
closed 2019-11-13 by raphaeltimbo