petrobras/ross #381: Update tutorial and examples

raphaeltimbo, 2019-11-18:
We had several API changes and we need to update the tutorial and examples to reflect that.

rodrigomoliveira1, 2019-11-18:
We could also release a new ROSS version to match the tutorial correctly. So, with each new version we have to release a tutorial update that conforms to the latest version.

Another idea is to make recurring updates in the tutorial (according to API changes and new features) but store them in another branch, then merging those changes **only when releasing a new version**, so we don't waste time reviewing every change from the last version to the new one.

raphaeltimbo, 2019-11-19:
> Another idea is to make recurring updates in the tutorial (according to API changes and new features) but store them in another branch, then merging those changes **only when releasing a new version**, so we don't waste time reviewing every change from the last version to the new one.

Agreed. I think we should also have some way of automatically run all notebooks after we deploy a new version. I think this is possible to do on travis CI:
```
before_deploy: "echo 'ready?'"
deploy:
  ..
after_deploy:
  - ./after_deploy_1.sh
  - ./after_deploy_2.sh
```
I will try to implement that.

rodrigomoliveira1, 2019-11-26:
I have finished updating all notebooks (examples and tutorial).
@GabrielBachmann is taking a look on them and will be adding them soon.

rodrigomoliveira1, 2019-11-28:
PR #395 closes this issue.

closed 2019-11-28 by rodrigomoliveira1