petrobras/ross #401: Let user choose material in "From section instantiation of a Rotor"

qwertz098, 2019-12-13:
Shaft elements default to "steel". Providing user's prefered material as kwarg forinit/with setter would ease things up.

qwertz098, 2019-12-13:
[image]

rodrigomoliveira1, 2019-12-13:
You are completely right. In addition, we could add this argument as a list if the user wishes add different materials in each section.

qwertz098, 2019-12-16:
Yes, a list would be fine. But please optionally, as most shafts (the main turned part without shrinked tubes and disks) consists of one material and that's easier for the user to specify.

rodrigomoliveira1, 2019-12-16:
@qwertz098,
We are implementing your request. I'll write down a simple example to help you out with the instantiation, since the tutorials still need to be updated. 
Also, when the PR is merged, you can download ross folder and replace the old one.

```
"""
        Parameters
        ----------
        leng_data : list
            List with the lengths of rotor regions.
        idl_data : list
            List with the inner diameters of rotor regions (Left Station).
        odl_data : list
            List with the outer diameters of rotor regions (Left Station).
        idr_data : list, optional
            List with the inner diameters of rotor regions (Right Station).
            Default is equal to idl_data (cylindrical element).
        odr_data : list, optional
            List with the outer diameters of rotor regions (Right Station).
            Default is equal to odl_data (cylindrical element).
        material_data : ross.material or list of ross.material
            Defines a single material for all sections or each section can be
            defined by a material individually.
        disk_data : dict, optional
            Dict holding disks datas.
            Example : disk_data=DiskElement.from_geometry(n=2,
                                                          material=steel,
                                                          width=0.07,
                                                          i_d=0,
                                                          o_d=0.28
                                                          )
            ***See 'disk_element.py' docstring for more information***
        brg_seal_data : dict, optional
            Dict holding lists of bearings and seals datas.
            Example : brg_seal_data=BearingElement(n=1, kxx=1e6, cxx=0,
                                                   kyy=1e6, cyy=0, kxy=0,
                                                   cxy=0, kyx=0, cyx=0)
            ***See 'bearing_seal_element.py' docstring for more information***
        nel_r : int, optional
            Number or elements per shaft region.
            Default is 1.
        n_eigen : int, optional
            Number of eigenvalues calculated by arpack.
            Default is 12.
        tag : str
            A tag for the rotor
"""
>>> from ross.materials import steel
>>> from bokeh.plotting import show
>>> rotor = Rotor.from_section(
...             leng_data=[0.5, 0.5, 0.5],
...             odl_data=[0.05, 0.07, 0.07],
...             idl_data=[0, 0, 0],
...             idr_data=[0, 0, 0],
...             odr_data=[0.07, 0.07, 0.05],
...             material_data=steel,
...             disk_data=[DiskElement.from_geometry(n=1, material=steel, width=0.07, i_d=0, o_d=0.28),
...                        DiskElement.from_geometry(n=2, material=steel, width=0.07, i_d=0, o_d=0.35)],
...             brg_seal_data=[BearingElement(n=0, kxx=1e6, cxx=0, kyy=1e6, cyy=0, kxy=0, cxy=0, kyx=0, cyx=0),
...                            BearingElement(n=3, kxx=1e6, cxx=0, kyy=1e6, cyy=0, kxy=0, cxy=0, kyx=0, cyx=0)],
...             nel_r=3)
>>> show(rotor.plot_rotor())
```

[image]

qwertz098, 2019-12-18:
Thanks a lot @rodrigomoliveira1 !

rodrigomoliveira1, 2019-12-18:
> Thanks a lot @rodrigomoliveira1 !

Thank you for the comments, @qwertz098 !

rodrigomoliveira1, 2020-01-18:
PR #405 closes this issue

closed 2020-01-18 by rodrigomoliveira1