petrobras/ross #402: Visualize geometrically defined disc elements geometrically

qwertz098, 2019-12-13:
[image]

[disk_element.py.txt](https://github.com/ross-rotordynamics/ross/files/3961428/disk_element.py.txt)

raphaeltimbo, 2019-12-16:
Hi there @qwertz098 !
Could you give more details on this issue?
If you could also point to the proposed changes to the disk_element.py file it would help us better understand the problem.

Thanks for taking your time to submit this issue!

qwertz098, 2019-12-17:
We have often to do with shrinking parts on a shaft, partly several layers, which i am going to define geometrically. (Pls see below, it's just a compromise)

As the geometry is know from the definition of the disk elements i assumed it's comfortable for the user to have it visualised properly, also for feedback to check if everything is entered fine. I took the source for the matplotlib patches from the shaft_element.py, added the dimension attributes for the disk element to the DiskElement class and added a case in the patch method in case the geometry is known. You can diff to v0.2 (master at the day of filing the issue).

There is a suggestion noted somewhere in the source to check the geometry for overlapping with other parts and in case to visualize for the user with a lower alpha blend while plotting that patch.

I did not implement any bokeh changes, as bokeh isn't working in my setup immediately (and don't wanted to invest any time).

The better solution for sleeves (in contrast to discs) would be the coaxial rotor issue (#277 ) which I see as critical for the progress of that useful tool here as it'd enable solving a lot of real world problems, at least in my world ;-) . But as this change was simple, i made it for me and just wanted to share, maybe it's of interest.

qwertz098, 2019-12-17:
I assumed, that if the geometrical data for the disc element is know (more a sleeve in my example), it might be usefully visualized using that data, for user feedback for example.

Therefore i used the matplotlib patch code from shaft_element.py, added gemtry attributes to DiscElement class and a matplotlib plotting case if the geometry data is availble.

You should see the changes by diff'ing to v0.2.

rodrigomoliveira1, 2019-12-17:
I understand your point, but taking a look to your coding suggestion, I don't think it's the best way to instantiate `DiskElement`.

You see:

```

class DiskElement(Element):
    def __init__(self, n, m, Id, Ip, tag=None, i_d=None, o_d=None, L=None):
        self.n = int(n)
        self.n_l = n
        self.n_r = n

        self.m = m
        self.Id = Id
        self.Ip = Ip
        self.tag = tag
        self.color = bokeh_colors[9]
        
        self.i_d = float(i_d)
        self.o_d = float(o_d)
        self.L = float(L)
```
It can get a little bit confusing to the user to enter mass and inertia data while having the option to enter geometry data at the same method. I think changing DiskElement.from_geometry would be better, look:

```
    @classmethod
    def from_geometry(cls, n, material, width, i_d, o_d, tag=None):
        m = 0.25 * material.rho * np.pi * width * (o_d ** 2 - i_d ** 2)
        # fmt: off
        Id = (
            0.015625 * material.rho * np.pi * width * (o_d ** 4 - i_d ** 4)
            + m * (width ** 2) / 12
        )
        # fmt: on
        Ip = 0.03125 * material.rho * np.pi * width * (o_d ** 4 - i_d ** 4)

        tag = tag
        cls.i_d = i_d
        cls.o_d = o_d
        cls.width = width

        return cls(n, m, Id, Ip, tag)
```

qwertz098, 2019-12-17:
Absolutely. I must admit not having thought about that at all, just wanted to see my plot asap. Not cleaning up the dirt.

rodrigomoliveira1, 2020-01-21:
@qwertz098, I believe you could add another Shaft Element overlapping other Shaft Element on the same node and setting a material with a Young's Modulus equals zero (only for the element that overlaps). It would have the same effect than adding disk elements

closed 2020-01-21 by rodrigomoliveira1