petrobras/ross #414: Bokeh version issue

raphaeltimbo, 2020-01-15:
Our code base is not working with older versions of bokeh due to the legend_label argument.
We need to define the bokeh version in our requirements or modify this part of the code to be backwards compatible.

rodrigomoliveira1, 2020-01-18:
Until release 0.1.7 we should define the bokeh version at 1.3.4.
From release 0.2.0 onwards, we should define the version of bokeh at 1.4.0

raphaeltimbo, 2020-01-21:
We should check how other packages manage their dependencies version.

LuanDaSilva, 2020-01-21:
After researching for a while, I figured out that developers in cases of bugs like ours force the software to update to the latest version. This is heavily used by companies like Microsoft and by the majority of game industry, that deal with several bugs with a short period of time. 

So, the quickest solution is to verify the user current version of ross and updating it to the latest one. After the update, a message recommends the user to read a 'what's new.txt' file generated by the program, informing the changes made.

LuanDaSilva, 2020-01-22:
In my newest research, I found that conda already have faced a problem similar to ours. The solution used by its developers have been the downgrade of some key packages. The whole discussion can be checked here: https://github.com/SciTools/cartopy/issues/853.

A problem that is commonly faced when some packages are downgraded  is the fact that people have been using them to other purposes besides runing ross. This type of question is solved by installing ross and its packages inside a virtual environment:https://uoa-eresearch.github.io/eresearch-cookbook/recipe/2014/11/26/python-virtual-env/ for python or https://docs.conda.io/projects/conda/en/latest/user-guide/tasks/manage-environments.html for conda

We should check wheter the user has conflict packages,and ,if does,we should recommend the instalation inside a virtual environment, to avoid package version conflicts.

raphaeltimbo, 2020-02-21:
Thanks for the research @LuanDaSilva .
This article [install_requires vs requirements files](https://packaging.python.org/discussions/install-requires-vs-requirements/) is also a good resource to understand how we can manage dependencies for our project.
In our case, we use install_requires inside the setup.py file and we should just set bokeh>=1.4.
If a user wishes to have an environment in which he can later reproduce some work using ROSS, he should locally maintain a requirements file for that environment.

raphaeltimbo, 2020-04-13:
I will update the requirements.txt file to solve this.

PR #537 (2020-04-27): Add bokeh required version
Closes #414.
commit 1806842c (2020-04-27): Add bokeh required version

Closes #414.
closed 2020-04-27 by rodrigomoliveira1