petrobras/ross #418: Improving the Tutorial - From section instantiation of a Rotor

rennergcm, 2020-01-17:
**It is hard to understand the explanation below, although it easier to create the rotor using this method(i_ds_data = [0,0,0]
o_ds_data = [0.05, 0.05, 0.05]
leng_data = [0.5, 0.5, 0.5]**.

n this form of instantiation, the number of shaft elements used in FEM are not fixed, instead, the program does a convergence analysis, testing the number of elements to a point where the relative error between iterations reaches a value that can be neglected.
To use this method, you should divide the rotor in a way where the number of shaft elements is minimal and place every element (except for the shaft elements) in the minimal nodes

raphaeltimbo, 2020-02-05:
@rodrigomoliveira1 , when we use the `.from_section()` method is the rotor automatically subdivided in several elements?

rodrigomoliveira1, 2020-02-05:
> @rodrigomoliveira1 , when we use the .from_section() method is the rotor automatically subdivided in several elements?

Yes, and the user chooses how many elements it should be inserted in each shaft section

commit c7b307e8 (2020-02-06): Improve from_section example

Closes #418.
closed 2020-02-06 by rodrigomoliveira1